cleanflight/cleanflight #2889: GPS does not work with v2.1.0-RC1

imarsv, 2017-07-14:
# A - Description

GPS does not work with v2.1.0-RC1. Failed to set 'gps_provider', 'gps_sbas_mode'.
Works well with v2.0.5 version.

# B - Steps To Repeat

Install 'Cleanflight / SPRACINGF3NEO 2.1.0 Jul  9 2017 / 20:50:51 (935dbdb)'
v2.1.0-RC1

# E - Version Information

Cleanflight: 2.1.0 Jul  9 2017 / 20:50:51 (935dbdb)
Configurator: 2.0.3

# E - Other information

FC: SP Racing F3 NEO

```
# feature GPS
unavailable

# set gps_provider = UBLOX
Invalid name

# set gps_sbas_mode = EGNOS
Invalid name

# diff
# Cleanflight / SPRACINGF3NEO 2.1.0 Jul  9 2017 / 20:50:51 (935dbdb)

feature GPS
profile 0

rateprofile 0

# status
System Uptime: 77 seconds
Voltage: 12 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=ICM20602, ACC=ICM20602, MAG=HMC5883
SD card: None inserted
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 3, config size: 2048, max available config: 4096
CPU:22%, cycle time: 136, GYRO rate: 7352, RX rate: 49, System rate: 9
Arming disable flags: 0x80

# tasks
00 - (         SYSTEM)      9       3       2    0.5%    0.5%         1
01 - (            PID)   1879      86      36   16.6%    7.2%     19220
   - (           GYRO)   7518
02 - (          ACCEL)    949      23      18    2.6%    2.2%      1264
03 - (       ATTITUDE)     99      46      35    0.9%    0.8%       269
04 - (             RX)     49     116      97    1.0%    0.9%       392
05 - (         SERIAL)    100    7227       3   72.7%    0.5%        87
07 - (BATTERY_VOLTAGE)     49       6       4    0.5%    0.5%        14
08 - (BATTERY_CURRENT)     49       4       2    0.5%    0.5%         5
09 - ( BATTERY_ALERTS)      4       5       3    0.5%    0.5%         1
10 - (         BEEPER)     99      11       2    0.6%    0.5%        16
11 - (        COMPASS)      9     110     108    0.5%    0.5%        84
15 - (      TELEMETRY)    249     287       5    7.6%    0.6%       180
16 - (       LEDSTRIP)     99     232     162    2.7%    2.1%      1111
17 - (    TRANSPONDER)    249       6       2    0.6%    0.5%        37
18 - (            OSD)     59    2514    1274   15.3%    8.0%      5948
19 - (            CMS)     59       5       2    0.5%    0.5%         9
20 - (        VTXCTRL)      4  350598       2  140.7%    0.5%       351
RX Check Function                   8       4                        17
Total (excluding SERIAL)                       191.6%   26.3%
```

McGiverGim, 2017-07-14:
In the actual code, GPS is only available for fc with more than 256Mb. If you need it you must create your custom build.

stanislavdavid, 2017-07-15:
I agree with @McGiverGim with some comments - limit is 256KB. 
If you want to make custom build - one possible solution is to edit `/src/main/target/common_fc_pre.h` - at the end there is disabled GPS and then folow other instructions in `/docs/development/`

imarsv, 2017-07-16:
Made custom build. Works. thx