cleanflight/cleanflight #2890: The motors #3 and #1 do not spin with DSHOT protocol on SP Racing F3 NEO

imarsv, 2017-07-14:
# A - Description

The motors #3 and #1 do not work (spin) with DSHOT (150/300/600) protocol on SP Racing F3 NEO

# B - Steps To Repeat

Set motor protocol DSHOT150.
Try to test motors with Cleanflight configuration.

# E - Version Information

Cleanflight: v2.1.0-RC1, v2.0.5
Configurator: v2.0.3

# E - Other information

```
# diff
# Cleanflight / SPRACINGF3NEO 2.1.0 Jul  9 2017 / 20:50:51 (935dbdb)

set motor_pwm_protocol = DSHOT150
profile 0

rateprofile 0
```

McGiverGim, 2017-07-14:
It works with oneshot? I've a similar problem, only multishot is working for me.

imarsv, 2017-07-16:
ONESHOT125 - works
ONESHOT42- works
MULTISHOT - works
DSHOT150 - #2 and #4 works only
DSHOT300 - #2 and #4 works only
DSHOT600 - #2 and #4 works only

Andi252, 2017-08-03:
Hi, I had the same problem and I think I found a workaround for Cleanfight 2.1.0:

1. load default settings in the configurator
2. enabled DSHOT600
3. disable LED_STRIPS
4. disable TRANSPONDER

with this configuration all 4 motors spin with DSHOT600

hydra, 2017-10-21:
yes that is correct, the transponder and LED strip must be disabled on the F3 NEO to allow the DMA resources to be used for DSHOT.

closed 2017-10-21 by hydra