cleanflight/cleanflight #2891: Newest Cleanflight PDF users manual? Where can I get it?

alanfj, 2017-07-14:
# 1. ARE YOU REQUESTING SUPPORT?

-- YES / NO --

# 2. If you answered YES to the above question go here instead:

https://www.rcgroups.com/forums/showthread.php?2249574-Cleanflight-firmware-for-STM32F3-based-FCBs-Check-First-Post-Please%21%21

Alternatively use other forums, slack or IRC for support.

# 3. If you answered NO to the first question.  Delete everything above this line, read the guidelines (see link above this form) and continue.

# A - Description

-- replace this line with a summary of the issue/problem/feature

# B - Steps To Repeat

-- replace this line with a list of steps to repeat your problem

# C - Video Links

-- provide a link to a video showing your problem, as applicable --

# D - Forum post links

-- provide links to forum posts if you have discussed this with others --

# E - Version Information

Cleanflight: vX.X.X or commit SHA
Configurator: vX.X.X or commit SHA

# E - Other information

-- provide additional information such as.
-- flight controller make/model
-- config dump + diff
-- esc make/model/config
-- receiver make/model/config
-- transmitter make/model/config
-- schematics
-- diagrams
-- operating system
-- etc.

hydra, 2017-07-17:
please don't use github for questions like this, use slack support.

answer: it's not been built yet but will be available for the final release and hopefully for the release candidates soon.

hydra, 2017-07-18:
further to this, the manual build tools no-longer work on the OS I use.

https://github.com/cleanflight/cleanflight/blob/master/docs/development/Building%20Manual.md

I created a PR to allow the manual to be built with markdown-pdf

https://github.com/cleanflight/cleanflight/pull/2897

markdown-pdf seems to be maintained and popular, but currently the hyperlinks in the generated PDF do not work, and there needs to be some styling work.

https://github.com/alanshaw/markdown-pdf

DanNixon, 2017-07-19:
@hydra If you want something a little more versatile you could have a look at [Pandoc](http://pandoc.org/).

hydra, 2020-05-17:
We can take a look at this if there is some interest.  Please create a new issue and reference this one if that's the case.

closed 2020-05-17 by hydra