cleanflight/cleanflight #2892: additional simple hardware support via "modes"

Hrumque, 2017-07-15:
Hardware control, RGB led WS2811b / WS2812

0. In flight modes there is an LEDLOW position - which completely shuts off all LEDs. Would it still be worth mentioning in the leds bar configuration tab? However, in the documentation https://github.com/cleanflight/cleanflight/blob/master/docs/Modes.md is still available mode LEDMAX and LLIGHTS which since version 2.x do not work

1. In the leds configurator - there is the "modes and orientation" option - but it does not cover all flight modes, which are available in "modes"
There should be a separate "modes" option, with either a choice or a mode list (so you can signal one led, eg Airmode or Antigravity mode, or simply control specific leds using a switch.

2a. (As a consequence of this, but also as a general extension) - it would be useful to create virtual (software) modes (like flight mode) that would be set in the Modes tab and could control the lighting scheme (activate leds that have this mode Selected), but also activate servo or other hardware.
It would be a smarter method than "servo forwarding" in a simple way, giving you much more options.
For example:
- If active failsave mode, turn on the loudspeaker and the flashlight
- If the battery is low, turn on the LEDLOW mode
- if the switch (eg on Aux4) is active, then enable LED4, LED5, LED6 (for front camera lighting), etc.

2b. Why in the mode configurator - you can select mode dependence from the AUX channel position and not from other parameters, ie 4 basic throttle / roll / pitch / yaw, battery level measurement, current consumption, measured altitude or speed etc As well as available variables in the software). After all, for example, it would not be foolish to be able to automatically activate a sonar (alt hold) mode less than 2m high, or turn off Acro (activate stabilize / angle) at low battery

3. It would be useful to create your own illumination diagrams - useful for planes so that the RGB leds could create lighting like a real plane (with the lights and the position lights blinking, strobe lights, etc). So to be able to define a glow, a flashing pattern, and be able to switch on / off a programmable switch ("ledlow" for the whole is a bit too much, for example GPS signaling is convenient to be permanently enabled)

4. It would be useful to simply control the GPIO pin of the processor.
In addition to the PWM PWM and servo pwm outputs, and other digital i2c usart, there are plenty of free processor pins that can be assigned to the "resource" by eg "DIGITALOUT" and controlled in on / off mode. For example, to control simple lighting (and high power, which in the case of WS2811 does not make sense), turn on additional water trombones - eg smoke generator, parachute trigger, camera switch, rocket firing, self-destruct;) etc. It could easily be assigned to Virtual modes, triggered

hydra, 2017-07-17:
airmode and antigravity are not real flight modes, they just improve flight behaviour, as opposed to the angle and horizon modes which correct pitch/roll and mag modes which correct yaw and headfree which changes the control method.

these are all good/interesting ideas though, thanks for sharing your thoughts.  

control of remote peripherals is usually done via pwm signals mapped to spare aux channels that are forwarded from the rx via the fc to the appropriate servo output.

i agree that better support for peripherals would be nice.

Hrumque, 2017-07-17:
> airmode and antigravity are not real flight modes, they just improve flight behaviour, as opposed to the angle and horizon modes
Yes I know. But someone might want to have signaling with the selected LED and the color "X mode is active" (whatever mode it would be, not just the few of the basic list).

hydra, 2017-10-21:
ok, understood.  Not something we currently have developer time for right now so closing.  Thanks for giving us the idea though, feedback always appreciated.

closed 2017-10-21 by hydra