cleanflight/cleanflight #2893: Camstab pins on Naze32 rev6 - no pin 7,8

Jvejvad, 2017-07-16:
# 3. If you answered NO to the first question.  Delete everything above this line, read the guidelines (see link above this form) and continue.

# A - Description

Not possible to use servo tilt/camstab on Naze32 rev.6

# B - Steps To Repeat

with servo tilt enabled - and camstab enabled. Its not possible to use servo pins.
Naze32 has 6 motor/servo pins - servo tilt / camstab requires pin 7 & 8.

# C - Video Links

-- provide a link to a video showing your problem, as applicable --

# D - Forum post links

-- provide links to forum posts if you have discussed this with others --

# E - Version Information

Cleanflight: v2.1.0 - 2.0.5
Configurator: v2.0.3

# E - Other information

-- provide additional information such as.
-- flight controller make/model
-- config dump + diff
-- esc make/model/cofig
-- receiver make/model/config
-- transmitter make/model/config
-- schematics
-- diagrams
-- operating system
-- etc.

hydra, 2017-07-17:
it doesn't require pins 7/8 remap moto outputs 1 and 2 to servo outputs, then remap motor output pins 3,4,5,6 to motor 1,2,3,4.

this used to happen automatically but now you need to use the cli `resource` command.

closed 2017-07-17 by hydra

Jvejvad, 2017-07-18:
How do i remap servos ?