cleanflight/cleanflight #2895: CLI ibata_scale does not accept negative value

AndreasAusD, 2017-07-16:
cleanflight 2.0.5, PIKOBLX board, configurator 2.0.3 

when trying to set ibata_scale to a negative value it is not accepted but allowed range is positive and negative

`# set ibata_scale = 1000
ibata_scale set to 1000
# set ibata_scale = -1000
Invalid value
Allowed range: -16000 - 16000

# 
`

[image]

matty17, 2017-07-17:
`# set ibata_scale = 1000 ibata_scale set to 1000
--

Invalid value Allowed range: -16000 - 16000
--

 
--

hydra, 2017-07-17:
please check with CF 2.1.0-RC2 and report back, i think this might be fixed already.

AndreasAusD, 2017-07-19:
yes, looks ok with 2.1.0-RC2

accepts the negative value and the current is displayed correctly in the configurator and by telemetrie

closed 2017-07-24 by AndreasAusD