cleanflight/cleanflight #2899: Wrong formula for current meter

McGiverGim, 2017-07-19:
Hi, I think that this PR is wrong: https://github.com/cleanflight/cleanflight/pull/2804
The scale, if I'm not wrong, is a divider of the sensor, not a multiplier: https://github.com/cleanflight/cleanflight/blob/f42c31b3f72c105d66221f7fc82a0aaef77f56ff/src/main/sensors/current.c#L112
@Schrolli91 @hydra what do you think? The original formula was ok for a divider of I'm not wrong. The corrected is right for a multiplier.

Schrolli91, 2017-07-20:
see here: https://github.com/martinbudden/betaflight/blob/master/docs/Battery.md
last lines....

The change in BF is not from me.
I have also tested the formula with my copter, and the formula from BF works well

McGiverGim, 2017-07-21:
The CF/BF changes are being copied from one to the other by the moment. So if you make changes in one of them, in some days the changes will be in the other.

If I'm not wrong, and the scale is a divider, like seen in the code:
```
internal_value / old_scale = reported_current -> internal_value = reported_current * old_scale
internal_value / new_scale = charging_current -> internal_value = charging_current * new_scale
```
Then:
```reported_current * old_scale = charging_current * new_scale```
Finally:
```new_scale = reported_current * old_scale / charging_current```
That is the same as:
```new_scale = old_scale / (charging_current / reported_current)```
that is what the old formula, before the change, was.

Can someone verify if my formulas are right?

There's one easy way to verify it: if the scale is bigger, the reported current will be smaller.

McGiverGim, 2017-07-28:
The scaler is a divider, not a multiplier:

https://github.com/cleanflight/cleanflight/blob/f42c31b3f72c105d66221f7fc82a0aaef77f56ff/src/main/sensors/current.c#L112

fiveangle, 2017-07-28:
I apologize, but I have not used the virtual sensor ever.  But I would have a hard time believing that you would need to lower the scaler to increase the virtual current draw reading.  Can you tell me  what you observe when you increase the scaler in BF ?  Does  Ahr  used during similar flight increase or decrease ?

McGiverGim, 2017-07-28:
I've made several tests. I needed to increment the scale to lower the final current. I've an Excel if you want to see it.

_Sent from my Google Nexus 5X using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

fiveangle, 2017-07-28:
If it decreases, then the real solution would be to fix the scaler to be more human friendly.  Perhaps this method was chosen to self limit the scaler (trading off resolution when scaler  approaches 0) ?

fiveangle, 2017-07-28:
If increasing the scaler = decreasing the reported current then you need to transpose the ratio elements and multiply the old scale (please see PR review comments). Regardless, manual calc should be presented as multiplying the old value by a ratio to get the new value. I'll leave it to you to figure out which value is the numerator and denominator :)

fiveangle, 2017-07-28:
Code says otherwise: https://github.com/cleanflight/cleanflight/blob/f42c31b3f72c105d66221f7fc82a0aaef77f56ff/src/main/sensors/current.c#L170

fiveangle, 2017-07-28:
You're confusing the ADC voltage scale with the virtual current sensor scale:

https://github.com/cleanflight/cleanflight/blob/4625646f31ce96ab29c8482621c6d0abe4e091a6/src/main/sensors/current.h#L67-L70

```
typedef struct currentSensorADCConfig_s {
    int16_t scale;              // scale the current sensor output voltage to milliamps. Value in 1/10th mV/A
    int16_t offset;            // offset of the current sensor in millivolt steps
} currentSensorADCConfig_t;
```

McGiverGim, 2017-07-28:
Yes, you're right. The tuning section can serve for both, but the real ADC is a divider and the virtual is a multiplier. I will close the PR and let decide if it's better to unify both our modify the doc with different formulas.
Thanks!

grizewald, 2017-09-25:
The formula presented in the documentation in Battery.md is certainly incorrect.

amperage_meter_scale = old_amperage_meter_scale * (charging_data_mAh / reported_draw_mAh)

should be:

amperage_meter_scale = old_amperage_meter_scale * (reported_draw_mAh / charging_data_mAh)

mikeller, 2017-09-25:
@grizewald: Can you please open a pull request to fix the documentation?

McGiverGim, 2017-09-25:
We have discussed it here: the problem is that the formula for the virtual current meter is not the same that one for the real current meter. One of them need the multiplier and the other the division.
Maybe it's better to unify then in code and use the same formula for both types. If not, we must put both in documentation. What is the better option?

mikeller, 2017-09-25:
The virtual current meter is a poor man's solution at best, so I would change that one to match the ADC way.

grizewald, 2017-09-25:
I'll make a pull request later this evening.

If would make sense if the documentation matches what the code does. As currently stated, the calculation in the documentation will increase the scale value when it should decrease it.

JRSHobby, 2017-09-25:
ok thanks

JRSHobby, 2017-09-25:
2899

fiveangle, 2017-09-26:
> @McGiverGim : We have discussed it here: the problem is that the formula for the virtual current meter is not the same that one for the real current meter. 

This is a solution in search of a problem.  The ADC calculation for `amperage_meter_scale` never introduces a second offline "adjustment" scaler, as the calculation is done with direct measurements and then the user sets `amperage_meter_scale` value based on plugging those measurements into the formula. That formula includes division because that's how ratios are calculated 😁 

For virtual sensor, they are set arbitrarily by the BF default of 400 (but the result is wrong because code has no true reference like the direct-measurment ADC method does).  User then works backward to determine `amperage_meter_scale` with the simple ratio calculation in the docs, arriving at the new value based on this offline "adjustment" ratio that is never actually used in code. 

Even if you transpose the formula elements to manipulate the formula from multiplying-a-ratio to dividing-an-inverse-ratio, it is still unrelated to the ADC measurement formula.

They are calculating different things.  Conflating them would more likely lead to confusion rather than clarity.

fiveangle, 2017-10-03:
Reviewing this discussion above, I see now there was ambiguousness of what was being referred to in various parts (whether to present scaler adjustment as multiplier or divide was being refered to, whether old BF linked Battery.md with formula one way or CF Battery.md which had formula another way was being referred to, or if adjustment calculation is inversely related to scaler or directly proportional was being referred to, etc).

While the reasons are manifold, bottom line is PR #2948 is still incorrect (and didn't fully change all the formulas anyway, so the page even contradicts itself between lines 161 and 170).

#2956 submitted to resolve, and I even went so far as to configure VCS on one of my quads and test the Battery Charger method so hopefully this will be the last time anyone has to deal with it.

hydra, 2017-10-21:
#2956 merged, closing this issue now.

closed 2017-10-21 by hydra