cleanflight/cleanflight #2900: 气压计MS5611代码的问题 problem in the code of barometer_ms5611.c

fanbinjim, 2017-07-20:
In cleanflight/src/main/drivers/barometer/barometer_ms5611.c 
Line 165 and Line 179

Start convert read temperature and pressure, but the SPI version has the same reg value, I think it is wrong.
start to read temperature
```
static void ms5611_start_ut(void)
{
#ifdef USE_BARO_SPI_MS5611
    ms5611SpiWriteCommand(CMD_ADC_CONV + CMD_ADC_D2 + ms5611_osr, 1); // D2 (temperature) conversion start!
#else
    i2cWrite(BARO_I2C_INSTANCE, MS5611_ADDR, CMD_ADC_CONV + CMD_ADC_D2 + ms5611_osr, 1); // D2 (temperature) conversion start!
#endif
}
```

start to read pressure
```
static void ms5611_start_up(void)
{
#ifdef USE_BARO_SPI_MS5611
    ms5611SpiWriteCommand(CMD_ADC_CONV + CMD_ADC_D2 + ms5611_osr, 1); // D2 (temperature) conversion start!
#else
    i2cWrite(BARO_I2C_INSTANCE, MS5611_ADDR, CMD_ADC_CONV + CMD_ADC_D1 + ms5611_osr, 1); // D1 (pressure) conversion start!
#endif
}
```

the SPI version havs the same CMD_ADC_D2 value, but the IIC version is different, CMD_ADC_D2 and CMD_ADC_D1, may be it is an oversight. 
这个地方的IIC的代码是对的，SPI的版本写成一样的了，可能是疏忽了吧。

fanbinjim, 2017-07-20:
another bug, in file /src/main/drivers/barometer/barometer_spi_ms5611.c
Line 49: 
```
spiTransferByte(MS5611_SPI_INSTANCE, reg | 0x80);
```
this read function unintentionally changes the ADC READ reg's high bit into 1, it caused the read to fail. 
Here is the datasheet of the ms5611. 
[image]

There are two types reading regs with ZERO in high bit or ONE in high bit. So, when we read the PROM Read, that is ok, it doesn't matter, but when come to the ADC Read, it fails.And I hope to help ^_^

hydra, 2017-10-21:
this looks to be fixed in master now, please re-open if this is not fixed.

closed 2017-10-21 by hydra