cleanflight/cleanflight #2906: Betaflight remnant on osd

ijustwannafly, 2017-07-27:
I flashed cleanflight 2.1 rc1 on my dys f4 aio using the omnibusf4 target.  It works for the most part except that the osd shows the betaflight logo when booted and other random bits of information that I cannot get rid of from the osd page in the configurator.

On betaflight I had to jump to 3.2 for sbus.  On cf 2.1 it works with sbus and everything else except the osd.

I tried flashing with and without full chip erase with no change.

Is the osd chip on this board reprogrammable or is the dys f4 aio board "doomed" to stay on betaflight?

McGiverGim, 2017-07-27:
You must upload the font for the OSD to get the CleanFlight logo, in the OSD tab.

ijustwannafly, 2017-07-27:
Ok I think I'm having a dain bramage moment.

I can't seem to find a font file that needs to be uploaded.

I can change fonts in the configurator to what is offered but as for uploading I can't seem to find any specific cleanflight font files.

Any insight?

Many thanks.

ijustwannafly, 2017-07-27:
Error on my part.  I needed to load the unpacked cleanflight configurator smh :facepalm:

No issues.

closed 2017-07-27 by ijustwannafly