cleanflight/cleanflight #2909: wrong GPS-Modes configuration

trekki1978, 2017-07-30:
Hi,
I tried to configure the GPS Hold mode in combination with BARO hold mode. Which didn't work at all.
Now I found out by checking the blackbox files that the mode GPS Hold in CF-Configurator results in PASSTHRU on the copter. Only when I configure the GPS Home mode I get GPS Hold. There must be a problem with the MSP IDs
[image]

The Beeper mode works fine.

Not sure if this is an issue with the configurator or with the firmware?!

Thank you
Trekki1978

McGiverGim, 2017-07-30:
With the blackbox viewer. Please use the latest version from github to test it.

trekki1978, 2017-07-31:
Hi, I forgot to tell that I'm using the SP Racing F4Evo board. Betaflight Blackbox Explorer Version 2.5.11 should be the latest version.
Now I tested only the Baro mode. But the copter is always overshooting. The Blackbox Explorer shows that the throttle stick is moving up and down although I don't touch it.
I will upload the blackbox file and the dump:

[LOG00002.zip](https://github.com/cleanflight/cleanflight/files/1186794/LOG00002.zip)
[dump_2017-07-31.txt](https://github.com/cleanflight/cleanflight/files/1186786/dump_2017-07-31.txt)

Please tell me if I shall open another issue.

Thanks
Trekki1978

hydra, 2017-10-21:
the CF BB configurator is likely out-of-sync with the latest firmware and is mis-reporting modes.  It needs work doing to it.  Try using the latest BB viewer or Betaflight BB viewer which is more up-to-date.  If there's still an issue please create a new issue and reference this one.

closed 2017-10-21 by hydra