cleanflight/cleanflight #2911: Logic for arming motors not logical

mmabln, 2017-07-31:
# A - Description

The logic to actually arm the motor(s) seems not be very logical to me (and to many other users apparently). Why is it required that the throttle signal is LESS than the minimum configured?

I would like to request to enable arming the motors when the throttle is LESS OR EQUAL to the minimum value. To me it does not make any sense that you have to set the travel on your transmitter to EXCEED the values configured in cleanflight.

The default settings are low 1100, high 1900 and center 1500. This fitted perfectly my rx/tx (Graupner HOTT SUMD) set at +/- 100% travel (TX default setting).

So it is irritating why I have to increase the negative throttle travel on the TX just to be ablte to arm the motors. 

Is this behavior documented somewhere?

# B - Steps To Repeat

Not able to arm motors with Grauper TX set to 100% travel and cleanflight at default settings (1100, 1500, 1900).

# E - Version Information

Cleanflight: v2.1.0
Configurator: v2.1.0
TX: Graupner MX.-12
RX: Graupner GR-12
Protocol: SUMD
FC: CC3D EVO

mmabln, 2017-08-02:
Okay, it is mentioned in the documentation, RTFM though. 

But when the user specified a mode to control the ARM function, using a dedicated switch on the TX, does it still make sense to require the throttle stick being LESS than the min value?

Say, as I do not plan to use any of the stick combinations, I would not like to set my travel to +/- 125% but leave it at +/- 100%. I would like to use dedicated switches instead and using the modes configuration.

However, I understand the ARM function is special for safety reasons. So if the user has specified a mode for ARM function - that means he most likely uses a switch on the TX for doing it - could it make sense to allow ARMing even with throttle EQUAL to minimum (maybe allow it to be between -99% and -100%), and not require it to be LESS than the minimum?

fftunes, 2017-08-05:
To be able to arm, throttle value has to be lower than min_check, not min_throttle or min_command.

You could increase min_check to 1110 or something like that.

hydra, 2017-08-17:
`min_check` is accepted as being needed to be higher than `min_throttle`.

`min_check` should always be set a little less than your stick travel range. that's how it's designed to be used.

what is your use-case for wanting to set them both to the same value?

fftunes, 2017-08-24:
> min_check is accepted as being needed to be higher than min_throttle.

Wrong. min_check needs to be higher than lowest throttle value from tx to allow arming.
min_throttle is fc->esc and has nothing to do with stick range.

hydra, 2017-10-21:
yes, my bad.  I was mis-remembering - part of the problem comes from the horrible names for the 'min_check' 'max_check' parameters which don't describe what they are used for...

the defaults for the min/max check values have to be set to allow most radios to work, there are some that need additional configuration, this is documented - historically and currently users are expected to set their channel endpoints to the requirements of the firmware.

I'm not saying we couldn't improve this situation by adding RX stick calibration into the firmware or configurator however, but that needs developer time to do it and will likely solve any such issues.

Closing for now, since there's not an actual bug.

closed 2017-10-21 by hydra