cleanflight/cleanflight #2915: NAZE firmware 2.1.0 magnetometer and gps not detected

carsax1, 2017-08-02:
Hi, 
I updated my FC (ACRO NAZE32 FULL ) with the latest version (2.1.0), 
no longer recognizes magnetometer and ublox 7 gps, you can help me thanks
The current Cleanflight version 2.1.0

hipher, 2017-08-02:
Hello

Same problem here. Just today I've decided to clean de dust from my quad with a NAZE32 full board, Last time I used it was long time ago, I was on cleanflight 1.6, so I've been catchig up with all the new stuff. But there was no way to get the MAG recognized after installing the latest firmware. Finally I tried with version 2.0.5 and it works (at least the configutator says so; I haven't flown MAG mode yet).

KenImhof, 2017-08-03:
I am also having this problem with the ublox 7 not be detected.  I replaced the ublox 7 with M8N and now the GPS is working; however, mag is not working with either the onboard mag on SP3 or external MAG on M8N with Omnibus F4.

fpvjonnyfive, 2017-08-03:
same issue here... i have a naze rev6 and trying to get the gps to detect with no luck. its a ublox 7 gps.

guidojones, 2017-08-03:
If you look in the code, they have disabled (not included) GPS support for f1 and f3 processors.
I would assume that if your mag was part of the GPS, it too wouldn't be recognized

carsax1, 2017-08-04:
Can I solve this problem? I need gps, is there a software version compatible with ublox7?

hugoget, 2017-08-08:
Hello, since the update 2.1 cleanflight can not use GPS modes on my Sp Racing f3 (Ublox Neo-7M). GPS modes are not included in the menu.

carsax1, 2017-08-08:
Hi, sorry I do not understand must be a generic firmware for any typology or mistake?

hugoget, 2017-08-08:
I use Cleanflight firmware 2.1.0 on my sp racing F3 card and since this update, the Ublox Neo-7M GPS does not remain configured correctly. At each restart it is enabled with the NMEA protocol, unable to activate Ublox. The GPS is therefore not usable in the modes therefore line gpshold nonexistent. In version 2.0.5 everything works.

guidojones, 2017-08-08:
All,
If you look in the source code, you will see that they have moved support for GPS from F3 targets to F4 targets due to memory constraints in the F1 and F3 chips. I haven't looked at each specific F1 / F3 FC target to see if GPS was added back in for it, but for now it will require a F4 or better FC for out of the box GPS support.

NAZE32 is an F1 chip. The developers of both Cleanflight and Betaflight have announced that by the end of the year they will no longer support F1 Flight Controllers.

If you want, there are great instructions provided in the docs on how to compile your own build enabling / disabling features you want. Doing this you could enable GPS support and remove support for other features you don't use / want.

fiveangle, 2017-08-08:
Other forks like iNav have gone even further and fully dropped Naze32 (and CC3D shortly) so if someone wants F1 to remain supported as before, they'll have to probably fork, or work on optimization of existing features to reduce their requirements then re-enable features for F1 targets.  Alternatively, a compile-time configuration mechanism where F1 users could enable/disable optional flash and mem-resident features as necessary so everyone still gets what they want is another option.

Marlin 3D printer FW ran up against this and kept going for years on older HW, but that was out of necessity since faster HW wasn't readily available: that issue doesn't exist here.

hydra, 2017-10-03:
nothing to do here, closing.

closed 2017-10-03 by hydra

nipunnawanjana1, 2018-02-20:
it work for me https://youtu.be/9Es1NzxLyWE

hbouzas, 2018-06-03:
Guys, see my blog for an explanation on how to solve this issue. Step by step. 
www.horaciobouzas.com