cleanflight/cleanflight #2918: Servo resource assignment

Merewether, 2017-08-09:
# A - Description
I am using the SP Racing F4 EVO flight controller in a fixed wing configuration. Servos are not properly assigned in either the AIRPLANE or CUSTOMAIRPLANE modes. Actual pin assignments do not match either the listing generated by the RESOURCE command nor the RESOURCE LIST command.

# B - Steps To Repeat
From loading DEFAULTS and allowing for a reboot, load lhe following script file in the CLI interface to verify the problem: 
[spr4evo_airplane_script.txt](https://github.com/cleanflight/cleanflight/files/1211828/spr4evo_airplane_script.txt)
Using this configuration, actual servo assignments are shifted where servo 3 is visible on the C06 port (assigned to servo 1) and repeats for all servos up to and including servo 8 which is visible on B07 (assigned to servo 6). This will work as a "work around" where I am driving the motor with the servo 3 output. Attempting to put the motor 1 on C06 (by swapping resource assignments for motor 1 and servo 3) breaks the servo assignments and I am unable to get the servos to function properly under any other resource mapping.

# C - Video Links
none

# D - Forum post links
I have worked with @mikeller on the slack channel (I am @kahn).

# E - Version Information

Cleanflight: v2.1.0

# E - Other information
-- provide additional information such as.
-- flight controller make/model: SPRACINGF4EVO
-- Firmware version: v2.1.0 - SPRACINGF4EVO - 30-07-2017 06:26 (stable)
-- config dump + diff
Dump:
dump
version
Cleanflight / SPRACINGF4EVO 2.1.0 Jul 30 2017 / 13:15:12 (8b24735)
name
name -
resources
resource BEEPER 1 NONE
resource MOTOR 1 A01
resource MOTOR 2 NONE
resource MOTOR 3 NONE
resource MOTOR 4 NONE
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 C06
resource SERVO 2 C07
resource SERVO 3 C09
resource SERVO 4 C08
resource SERVO 5 B06
resource SERVO 6 B07
resource SERVO 7 B01
resource SERVO 8 B00
resource PPM 1 A03
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
[... 684 lines cut ...]
rxrange

vtx

rxfail

master
set min_throttle = 1000
set align_board_yaw = -90
set current_meter = NONE
set disarm_kill_switch = OFF

profile
profile 0

rateprofile
rateprofile 0

-- esc make/model/config
-- receiver make/model/config
-- transmitter make/model/config
-- schematics
-- diagrams
-- operating system
-- etc.

hydra, 2017-10-21:
please build firmware from latest master and re-test.

hydra, 2018-02-22:
please try 2.2.0-RC2 and report back.

hydra, 2020-05-17:
no feedback report and `master` has changed significantly again.  closing as Inactive.

closed 2020-05-17 by hydra