cleanflight/cleanflight #2919: Servo filter documentation needs updating

Merewether, 2017-08-09:
# A - Description
According to the current documentation on Github, there are two variables to filter the servos; servo_lowpass_freq and servo_lowpass_enable (https://github.com/cleanflight/cleanflight/blob/master/docs/Mixer.md#servo-filtering). When I do a 'get servo' command in the CLI, the only variable that seems to represent servo filtering is: 'servo_lowpass_hz' and the set function returns the error 'invalid name' when trying to 'set servo_lowpass_enable = ON'. Looking at the code, having a non zero value in the 'servo_lowpass_hz' variable is all that is required to start the feature. Just a note to update the documentation to describe this updated method of servo filtering initialization.

hydra, 2017-08-17:
much of the documentation needs updating, you can help by checking out the documentation and updating it if you see things that are wrong and/or out of date.  Then submit a PR from your changes and we'll merge the updates into the documentation.

hydra, 2017-10-21:
closing, no PR received.

closed 2017-10-21 by hydra