cleanflight/cleanflight #2931: SPF3 6DOF external mag not connected

matevz25, 2017-08-28:
Hello my SPF3 6DOF external mag HMC5883L doesnt work. Its connected to I2c port. Cleanflight wont recognize it. CLi mag enabled HMC5883, tried 3,3V power  from FC and  5V from PDB. Same. Scl and Sda have aprox 2,2V input/ output. I have tried 2 sensors. I have tried i2c highspeed but command doesnt work. 
What else can I do? FW is 2.0.5, on latest gps doesnt work for me.Thank you !

mikeller, 2017-08-28:
GPS has been disabled on F3 because of flash size constraints.

matevz25, 2017-08-29:
Ok, on 2.0.5 gps works. Problem is mag. Why it doesnt work?

hydra, 2017-10-03:
if your firmware has mag support then mag will work on the SPRacingF3 since the other sensors are also connected via I2C.   CLI commands for mag would not be available if there was no mag support in the firmware.

I don't thing we have enough information to solve your problem. Closing.

closed 2017-10-03 by hydra