cleanflight/cleanflight #2946: CleanFlight Passthrough Not supported Thru BLHeli Suite

ConfusedMind15, 2017-09-23:
# A - Description

I'm not able to connect through to BLHeli Suite via CleanFlight. The error message says My current cleanflight configurator is not supported via BLHeli Suite

# B - Steps To Repeat

-- replace this line with a list of steps to repeat your problem

# C - Video Links
[image]

# D - Forum post links

# E - Version Information

Cleanflight Configurator: v2.1.0

# E - Other information

FC:
-- Naze32 Rev.6 

ESC: 
-- 3 Turnigy 10A BLHeli 2-4S (Opto)
-- 1 Turnigy 10A Plush with BEC

FlySky FSiA6B RX 
Flysky FSi6S TX

mikeller, 2017-09-23:
ESC passthrough has had to be removed from F1 targets to make the firmware fit into the limited flash space available.

Workaround: Backup your config in CLI (`diff`, then 'Save'), flash an older version of cleanflight, use it to flash / configure your ESCs, flash newest version of cleanflight, restore your config by copy / pasting it into CLI.

JRSHobby, 2017-09-24:
I have not been able to set. SP F3-PDB-I am not able to set up UAVFeatures-99-Build I need to learn how or ? I do Have* another *Eachine  X220S OmnibusF4-5.8G-72CH.VTX-30A-Dshot600-2206-2300KV-800TVL. Were should I Set up at

ConfusedMind15, 2017-09-25:
@mikeller Thanks a lot  
I saved my current config, flashed an older version of cleanflight and was able to access BLHeli Suite.

JRSHobby, 2017-09-29:
I am like walking on glass without shoes. newbee trying feeling sad.

hydra, 2017-10-03:
thanks for the update.

closed 2017-10-03 by hydra