cleanflight/cleanflight #2950: Cannot run program "": Launching failed  Error: Program "" not found in PATH

helpmelearnplease, 2017-09-27:
Hello,
I am trying to build the cleanflight code in Eclipse by following the steps in the link
https://github.com/cleanflight/cleanflight/blob/master/docs/development/Building%20in%20Eclipse.md
and i am not able to build the program after importing the same.
I am getting the error as mentioned below,

Cannot run program "": Launching failed

Error: Program "" not found in PATH
PATH=[C:\Program Files\GNU ARM Eclipse\Build Tools\2.8-201611221915\bin;C:/Program Files/Java/jre1.8.0_131/bin/server;C:/Program Files/Java/jre1.8.0_131/bin;C:/Program Files/Java/jre1.8.0_131/lib/amd64;C:\ProgramData\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\eclipse;C:\MinGW\msys\1.0\bin]
Kindly help me out in this.
My operating system is Windpws7

mikeller, 2017-09-27:
Looks like you've missed some of the steps shown in https://github.com/cleanflight/cleanflight/blob/master/docs/development/Building%20in%20Eclipse.md. 

I would recommend that you first make sure that this works: https://github.com/cleanflight/cleanflight/blob/master/docs/development/Building%20in%20Windows.md

After that, move onto building using Eclipse.

closed 2017-10-03 by hydra

helpmelearnplease, 2017-10-03:
@mikeller I tried the CyGwin but there are some packages that is not available as listed in the documentation.
Kindly help me sort this out

mikeller, 2017-10-05:
@helpmelearnplease: I can't help you much there, I am using linux.