cleanflight/cleanflight #2951: QX95 beeps continuously in flight with 2.1.0 but not if I downgrade back to 2.0.5

george-hawkins, 2017-09-27:
# A - Description

I upgraded a new [Eachine QX95](https://www.banggood.com/Eachine-Tiny-QX95-95mm-Micro-FPV-LED-Racing-Quadcopter-Based-On-F3-EVO-Brushed-Flight-Controller-p-1085818.html) to 2.1.0 and added a buzzer. I used a parameter set that had worked perfectly with a previous QX95 running 2.0.X versions of Cleanflight.

When I took it out for its first flight I found that it beeped continuously during flight - the beeping was similar to the armed-but-no-throttle beeping sound, i.e. not as fast as the lost copter sound. It was very slightly more irregular than the armed-but-no-throttle sound but still fairly regular, i.e. it wasn't just erratic occasional beeping. I assumed that I'd got something wrong when connecting up the buzzer (a [Matek WS2818B](https://www.banggood.com/Matek-WS2812B-LED-Board-With-5V-Buzzer-For-Naze-32-Skyline-32-Flight-Controller-p-991715.html?rmmds=searchhttps://www.banggood.com/Matek-WS2812B-LED-Board-With-5V-Buzzer-For-Naze-32-Skyline-32-Flight-Controller-p-991715.html) combined buzzer and LED strip) so I took it home and checked everything but couldn't find any issue.

As I'd had the same setup working without issue on another QX95 running Cleanflight 2.0.X I eventually downgraded to 2.0.5 and this resolved the problem. So the beeping seems to be the result of some change introduced in 2.1.0.

# B - Steps To Repeat

Arm the QX95 - apply throttle, once the the propellers start spinning the armed-but-no-throttle beeping stops but once enough throttle is applied that the craft takes off a very similar but slightly more erratic beeping starts. The craft flies fine - but the continuous beeping drives one to land it and stop.

# C - Video Links

# D - Forum post links

# E - Version Information

Cleanflight: v2.1.0 or commit SHA
Configurator: v2.1.0 or commit SHA

# E - Other information

I've attached the output produced by `diff` for the 2.1.0 setup I was using - see [parameters.txt](https://github.com/cleanflight/cleanflight/files/1338832/parameters.txt). This is same configuration that I used when I downgraded to 2.0.5 - actually I downgraded, entered `defaults` and then applied this set of parameters (though `rc_expo_yaw` was called `rc_yaw_expo` prior to 2.1.0 - so I had to change that).

I was using Cleanflight / SPRACINGF3EVO 2.1.0 with the [FRF3_EVO_BRUSHED Flight Controller](https://www.banggood.com/FRF3_EVO_BRUSHED-Flight-Controller-Built-in-Frsky-8CH-Sbus-Receiver-For-Eachine-QX95-QX90-QX90C-p-1105907.html).

I'm not very familiar with Cleanflight - but I could reproduce this situation while tethered - so while the QX95 has no SD card I could capture logged data while tethered if this would be useful and you told me what items to tick for data capture?

mikeller, 2017-09-27:
This sounds a lot like it's a low battery warning. Can you check, when connected over USB and plugging in a battery, that the correct number of cells is detected? (Battery tab)

JRSHobby, 2017-09-28:
yes

JRSHobby, 2017-09-28:
conected but boad is pointing wrong* direc*

JRSHobby, 2017-09-28:
Iwill check setting

JRSHobby, 2017-09-28:
[en.DM00218346Clenflight.config.pdf](https://github.com/cleanflight/cleanflight/files/1339093/en.DM00218346Clenflight.config.pdf)

JRSHobby, 2017-09-28:
Motors spin  *RX* TX * Not Talking to Receaver-switch setings or

JRSHobby, 2017-09-28:
Nothing works for receaver motors spin in cleanflight  but reaceaver no readings of throtle exet,

george-hawkins, 2017-09-28:
Thanks @mikeller for responding within minutes of me logging the issue - I went to bed after logging it so I've only had a chance now to look at things. Your suspicions were correct.

In 2.0.5 Clearflight couldn't detect the presence of the battery at all with this particular board:
[image]

But with 2.1.0 the battery is detected and at rest all looks good:

[image]

But on applying enough throttle to take off the reported voltage immediately drops from the 3.7V seen above to 3.4V and so the battery warning kicks in. The batteries seem in good condition and the craft can fly for the amount of time I expect so I just adjusted the voltage warning down.

I'm not sure what to make of the comments from @JRSHobby - this is either bizarre spam or I guess the person in question has accidentally followed up here on something that's part of another issue.

closed 2017-09-28 by george-hawkins