cleanflight/cleanflight #2955: No SoftSerial ports defined for target FRSKYF4

benderl, 2017-10-02:
# Description
No SoftSerial ports defined for target FRSKYF4. I suggest to use motor pins 5+6 as rx/tx for quad builds.

# Steps To Repeat
Enable the feature SoftSerial in the configuration and reboot the FC. The feature is active but no SoftSerial ports appear in the ports tab.

# Version Information
Cleanflight: v2.1.0
Configurator: v2.1.0

# Other information

- flight controller make/model: FrSky XSRF4O

McGiverGim, 2017-10-02:
If I'm not wrong, you must add the pins you want to use with the "resource" command.

benderl, 2017-10-02:
You are absolutely right! I was not aware of this feature.
What I did on the CLI:
```resource motor 5 none
resource motor 6 none
resource serial_tx 11 a1
resource serial_rx 11 a8
save
```
Thanks!

closed 2017-10-02 by benderl