cleanflight/cleanflight #2963: limit the maximum throttle on anti turtle mode

grhookgr, 2017-10-14:
sometimes we push the stick too much making the copter to jump to much, = roll 360.
it will be nice if we have the option to limit the maximum throttle on anti-turtle mode.

Jahnkeanater, 2017-10-14:
I would rather have too much throttle than not enough. Some quads are more difficult to flip than others. If there was a limit it would need to be configurable.

fiveangle, 2017-10-14:
Do you have expo on your TX ?

grhookgr, 2017-10-14:
i was thinking a "set max_recover_throttle= 1750"  or the desired value it may just work to help us recover faster. 
it may be even configurable from osd

fiveangle, 2017-10-14:
Is that a yes or a no ? 😁

grhookgr, 2017-10-14:
:) No expo

Quick-Flash, 2020-06-27:
this has been implemented into EmuFlight

haslinghuis, 2020-07-07:
Cleanflight >= 4.2.0 has [new rates modes](https://github.com/cleanflight/cleanflight/wiki/4.2-Tuning-Notes#new-rates-modes) :)
I guess rates and expo are a personal preference.

closed 2023-04-19 by hydra