cleanflight/cleanflight #2964: Can i use uart1 to connect bluetooth for control Naze32 via MSP?

sorawitt, 2017-10-15:
i want to controll Naze32 to fly via MSP by bluetooth.

mikeller, 2017-10-15:
@sorawitt:
1. Yes this is possible;
2. RX_MSP is intended for development and bench testing, not for in-flight control. It does not have the same level of built in safety as the other RX protocols (failsafe,...). Using it for in-flight control without additional safety measures is potentially dangerous.

sorawitt, 2017-10-15:
@mikeller 
i send command[MSP_SET_RAW_RC] to Naze32 but nothing happen!
I just want to send command with MSP because I do not have radio transmitter and receiver.

mikeller, 2017-10-15:
You'll have to build your own custom firmware, since RX_MSP is not built into the released version for NAZE, due to flash space constraints.
Also, you should really get yourself a TX and RX, before you give somebody a bloody face with your craft.

closed 2017-10-21 by hydra