cleanflight/cleanflight #2966: Ublox GPS Module losing u-center settings once plugged into f3 acro flight controller

jigglemein, 2017-10-18:
# A - Description

I was hoping to get some help with an issue with setting up my u-blox gps with cleanflight. 
I have a GY-NEO6MV2 gps module and updated the configuration in u-centre 8.27 as per the instructions at: https://github.com/cleanflight/cleanflight/blob/master/docs/Gps.md.

The settings where successful and plugging the gps in and out from the ftdi module the configuration stays in place. The problem is when in cleanflight I follow the instructions but the green GPS module does not activate.

When I plug the gps back into u-centre the configurations are back to default. Port is back to 9600 and the rest of the configurations are lost. It seems to reprogram the GPS as soon as its plugged into uart2.

Instructions I have been following:
https://www.youtube.com/watch?v=8FIi_xuH4Vo

# B - Steps To Repeat
1) Update GPS with u-centre and save config as per https://github.com/cleanflight/cleanflight/blob/master/docs/Gps.md
2) Check its now connecting 57600
3) Plug GPS into f3 acro on uart2
4) Enable uart2 for gps on 57000, save and reboot
5) Go to config and enable gps with ublox config
6) Save and reboot
7) GPS does not activate
8) Plug back into u-center
9) Setting all back to default.

# E - Version Information

Cleanflight: SPRACINGF3 2.1.0
Configurator: 2.1.0 

# E - Other information

F3 Flight Controller Acro 6 DOF

mikeller, 2017-10-18:
@jigglemein: The GPS does not work on your flight controller because the GPS functionality was removed from the firmware for all F3 based boards to make the firmware fit into the limited amount of flash space.

If you want to use GPS with your flight controller, you'll have to build a custom version of the firmware that has built in support for GPS (and drops some other functionality to make space for it).

jigglemein, 2017-10-18:
@mikeller, thanks Michael.
Would you have a suggestion of a flight controller that does have the GPS functionality?

mikeller, 2017-10-18:
The SPRACINGF4EVO (http://seriouslypro.com/spracingf4evo) seems to be a good choice, and it has excellent support for cleanflight.

jigglemein, 2017-10-18:
@mikeller thanks, ordered. Closing ticket.

closed 2017-10-18 by jigglemein