cleanflight/cleanflight #2975: Smartaudio with TBS Unify Pro 5G8 HV Race Edition not working SPRF4EVO

Mmarsian, 2017-10-28:
# A - Description

Smartaudio with TBS Unify Pro 5G8 HV Race Edition not working on newest Cleanflight 2.1.0 and SP Racing F4 EVO stacked with SP Racing F3 OSD/PDB.
On same system flashed with Betaflight 3.2.1 it works immediately.

# B - Steps To Repeat

Install Cleanflight 2.1.0 on SPRF4EVO+F3OSD/PDB Smartaudio doesnt work. Statistics in OSD are all zero.
Then flash Betaflight 3.2.1 on the same system and it works like a charm.

# 

Cleanflight: v2.1.0  vs. Betaflight 3.2.1
Configurator: v2.1.0 vs. Betaflight 3.2.2

# E - Other information

-- SP Racing F4 EVO (original)
-- T-Motor AIR 20A
-- FrSky XSR
-- Taranis X9D+
-- used straight cable or voltage divider (both work with Betaflight)

# Dump 

# version
# Cleanflight / SPRACINGF4EVO 2.1.0 Jul 30 2017 / 13:15:12 (8b24735)

# name
name CrossFly

# resources
resource BEEPER 1 C15
resource MOTOR 1 C06
resource MOTOR 2 C07
resource MOTOR 3 C09
resource MOTOR 4 C08
resource MOTOR 5 NONE
resource MOTOR 6 B07
resource MOTOR 7 B01
resource MOTOR 8 B00
resource MOTOR 9 B10
resource MOTOR 10 B11
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 B06
resource SERVO 2 A10
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A03
resource PWM 2 A02
resource PWM 3 A09
resource PWM 4 A10
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 C10
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
[... 799 lines cut ...]
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

# rateprofile
rateprofile 2

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

mikeller, 2017-10-28:
This sounds like the issue that has been reported in betaflight/betaflight#3692.

RealNightEagle, 2017-11-15:
I tried this to fix the same problem. It didn't work, are there any other ideas?

closed 2017-11-16 by mikeller