cleanflight/cleanflight #2979: v2.1.0: SPRACINGF3OSD v2.1.0 hex is wrong, and Configurator does not work on Linux

Lecostarius, 2017-11-01:
# A - Description

This issue describes actually two errors.
1. The precompiled hex file of the 2.1.0 version of the target SPRACINGF3OSD does not work. Actually, I believe that it is compiled from the 2.1.0 version of the source code, but from a slightly outdated version. As a result, there will be no overlay if a camera is connected - this is an issue with the initialization of the OSD board.
2. The 2.1.0 version of the configurator has an additional comma in Features.js which has already been mentioned in a pull request (#503) for Configurator, but not pulled into master. It results in a Configurator that has only the CLI tab available and nothing else. The Windows version of 2.1.0 works fine. After removing the , as seen below, the 2.1.0 configurator works fine under Linux (I use Ubuntu 16.04), too.

         if (semver.gte(CONFIG.apiVersion, "1.20.0")) {
             features.push(
-                {bit: 18, group: 'other', name: 'OSD'},
+                {bit: 18, group: 'other', name: 'OSD'}
             );
         }

# B - Steps To Repeat

1. Flash the board with the official version of SPRACINGF3OSD target v2.1.0 (dated 30-07-2017 14:26) using the Configurator as usual. After flashing the SPRACINGF3EVO (sorry, only tried that) and a camera, there is no overlay although everything is configured properly. Actually, if you upload a new font to the PDB board while it and the EVO are both powered up by Lipo, the overlay picture will appear; however if you disconnect power and power up again, the overlay will be gone.

2. Use 2.1.0 with a 2.1.0 flashed board using chromium-browser under Linux. No graphical configuration is possible, only the CLI tab exists.

# E - Version Information

Cleanflight: v2.1.0 precompiled hex (using Configurator)
Configurator: v2.1.0. 

# F - Other information

1. I tried to compile 2.1.0 for the SPRACINGF3OSD from source and flashed it. With this, everything works. Therefore, I believe that accidentally a wrong version of the compiled hex file of the 2.1.0 version for SPRACINGF3OSD is supplied, which could be changed.

2. I removed the , from file "Features.js" as suggested in pull request #503 and after that, everything works fine.

hydra, 2017-11-03:
do you mean 2.2.0 or 2.1.0?  If you mean 2.1.0 that I can assure you it works, I flashed it the other day to a couple of boards.  Remember to follow instructions in the release notes for OSD slave boards.  Don't forget to enable OSD feature on FC too.

hydra, 2017-11-03:
also, please describe one problem per issue, it makes the conversation impossible to follow otherwise.  there's already the issue raised in the configurator issue tracker so no need for duplication here.

Lecostarius, 2017-11-03:
OK, sorry, so I will not talk about the Configurator Problem any more, as you say, it is well documented and just needs someone to accept the pull request 503.
As for the 2.1.0 of SPRACINGF3OSD, yes, I am talking about 2.1.0, not 2.2.0. 
I am pretty sure I am configuring it right since I can get it to work if I compile it from (2.1.0) source, but if I am loading the precompiled hex file, it does not work. In order to reproduce it you Need to power down the board and power up again, after that, a OSD flashed with the precompiled hex file will only Show the camera Picture and no overlay. but a OSD flashed with a newly compiled hex file works without problems.

hydra, 2018-03-07:
Unable to reproduce.  The units I have work fine here using the official releases.  Not sure what firmware you were using on your FC, I was using CF for my tests.  In BF there have been some changes which may have broken it for you if that is the case.

Please also see #3028 which I just merged which should help a little too.  Please re-test a nightly CF 2.3.0 that includes the PR or build it yourself (Flash BOTH the OSD SLAVE and the FC).

closed 2018-03-07 by hydra