cleanflight/cleanflight #2981: Request: Enable or disable telemetry via switch

ijustwannafly, 2017-11-12:
I have looked for a while now.

Old resources showed telemetry_switch option to enable/disable telemetry if it was used on the same uart as msp.

It would be nice if telemetry could be enabled or disabled at will via switch. I am talking about the telemetry slider in the config page.  

I found old resources that listed an ID number for the telemetry option as being 19 or 20.  If I can at least map it via cli to an aux channel using the aux command that would work too.

If this is a dumb single use request I understand.

Many thank.

mikeller, 2017-11-12:
@ijustwannafly: This option is still around, just that it's called `tlm_switch` now. If you set this on in CLI you will get a `TELEMETRY` mode that you can assign to a switch, and then use that switch to switch between telemetry and MSP.

ijustwannafly, 2017-11-12:
Thanks mikeller.

I have one setup using telemetry through softserial so it's not on the same port as msp.  Would it still toggle telemetry on and off in this scenario?

Currently I only need it on this rig while disarmed to change vtx settings.

mikeller, 2017-11-12:
I don't quite see why you would want to switch telemetry on and off in this scenario, and not just leave it on. Also, I don't know if the telemetry switch will work if the port is not shared, since this will probably prevent the port from being initialised at boot time.

ijustwannafly, 2017-11-13:
Mikeller I agree.

In fact even with the minor cpu overhead of using smartport over softserial it actually doesn't affect my rig.  It was just questioned to me about using smartport over softserial and the cpu load.  My rigs don't have acc enabled and the f4 boards have not given me issues at all.

It really comes down to my second guessing myself and my rigs due to someone else questioning my setup.

closed 2017-11-13 by ijustwannafly