cleanflight/cleanflight #2986: define new target

nyway, 2017-12-14:
Hi
According our demand we designed a type of FC hardware PCB , the interfaces definition are different with the current type of hardware , so we hope to customize a type of FC . Customizing a name for FC and then people could find and upgrade this FC in the betaflight firmware flasher list, For example we want to make a type of FC hardware which will be named with XXXX405. 
My question is how could we acchieve that and what procedure does it take ?

McGiverGim, 2017-12-14:
I'm not too sure what you want. But if you have the FC, and this FC will be available, you only need to made the modifications in the code and push a PR with your changes. If the code is ok, then it will be merged. If not, the developers will ask for changes.

If you have doubts, you can go to the slack chat and ask there too.

Only one thing more: you have post this message into cleanflight, but your message references betaflight ;-) Maybe you can ask there too.

mikeller, 2017-12-14:
Also, please note that cleanflight is pulling in changes from Betaflight, so a pull request for a new target should only be opened with one of the two, but not twice for cleanflight and Betaflight.

nyway, 2017-12-14:
@McGiverGim 
Thank you for your response，I should learn more English，then I can communicate better^-^。
I have a question,What is "push a PR",I can't unsderstand.
My FC changed complete,just some ports and anothet define

McGiverGim, 2017-12-14:
You need to read a bit about how git and github work. You must do a fork of cleanflight or betaflight. Then create a branch with your changes in your fork. And when ready, you must do a pull request against cleanflight or betaflight.

hydra, 2018-02-22:
yes, that's the procedure. closing.

closed 2018-02-22 by hydra