cleanflight/cleanflight #2988: any port possi

KiranKanchi, 2017-12-15:
# 1. ARE YOU REQUESTING SUPPORT?

-NO --

# 3. If you answered NO to the first question.  Delete everything above this line, read the guidelines (see link above this form) and continue.

# A - Description

Anybody tried FCU001V1 flight controller evaluation board from ST ? 
if yes, any port planned ? 

http://www.st.com/en/evaluation-tools/steval-fcu001v1.html

# C - Video Links

https://www.youtube.com/watch?v=vGFqIDVpMT4

# D - Forum post links

https://github.com/STMicroelectronics-CentralLabs/ST_Drone_FCU_F401

mikeller, 2017-12-15:
@KiranKanchi: Good suggestion!

hydra, 2018-09-28:
no-one has picked up on this, closing as there appears to be no interest in the target.

closed 2018-09-28 by hydra