cleanflight/cleanflight #2992: Is there away to dump out the Serial Input

HardVeur, 2017-12-28:
I try to control my CC3D with Cleanflight over MSP my Input seems to be ok but I can’t check it because it doesn’t seems like CF provides a way to check the serial rx MSP input so how can I figure out if the controller accept/recognizes my input correctly?

mikeller, 2017-12-28:
You should be able to see the bars moving in the 'Receiver' tab of the configurator.
(Also, for this to work, you will have to build your own firmware, since the MSP RX has been removed from the CC3D due to lack of flash space.

HardVeur, 2017-12-28:
I flashed the 2.2.0 version of CF on my cc3d and there is still a tab to enable serial rx as receiver input in the configurator but you say it doesn’t work with serial rx?

mikeller, 2017-12-28:
@HardVeur: No, I said it does not work with MSP RX. And yes, it is still in the list, since the configurator is not equipped to detect if a feature has been removed from an individual target.

HardVeur, 2017-12-29:
Which is the latest version of CF which does support MSP?

mikeller, 2017-12-29:
All versions support MSP. I think what you wanted to ask is what versions support MSP_RX. In this case, the answer is 'all versions support MSP_RX, but it has been removed from F1 targets like the CC3D from version 2 on. I am not sure if it was supported in for F1 targets in version 1, but you can try.

HardVeur, 2017-12-30:
Ok thanks for your answers i really appreciate it :)

I wanna explain my problem: my goal is to control my CC3D with MSP from a raspberry pi. It doesn’t matter if I control my CC3D over USB VDC MSP or over the main Port with serial rx MSP(gpio pins). Now is my problem that I already tried to connect my CC3D over USB and send MSP data to control throttle, yaw, pitch and roll just like a controller. But it didn’t worked. So now I have two problems: 

1) I don’t really understand the different between what serial rx MSP and USB MSP is(you maybe already noticed)?

2) And which options do I have to enable in Cleanflight to enable the MSP control of the motors over a USB port or either over UART1(I think MainPort) or UART2(I think FlexPort)?

mikeller, 2017-12-30:
1. MSP is the protocol that the configurator (and other apps) uses (normally over USB) to configure the firmware and get live data. RX_MSP is a receiver protocol implementation that uses the same format as MSP;
2. There are 3 ways I can see:
- get yourself a newer flight controller. Most flight controllers based on F3 or better support RX_MSP with standard firmware;
- build a version of the firmware for CC3D with RX_MSP enabled. See https://github.com/cleanflight/cleanflight/blob/master/docs/Customized%20Version.md ;
- instead of RX_MSP, use a serial protocol that is supported by standard firmware for CC3D, like SUMD. There are implementations for SUMD out there that you could use: https://github.com/openLRSng/openLRSng/blob/4787f92ab677e81749b7f569c1185b6bdeba4715/serialPPM.h#L95-L111

hydra, 2018-03-07:
closing, no updates.

closed 2018-03-07 by hydra