cleanflight/cleanflight #2995: Sonar doesn't work on STM32F3Discovery

emmhaych, 2018-01-05:
**SOLUTION PROVIDED BELOW**

# 1. ARE YOU REQUESTING SUPPORT?

NO

# 3. If you answered NO to the first question.  Delete everything above this line, read the guidelines (see link above this form) and continue.

# A - Description

Sonar sensor doesn't work on STM32F3DISCOVERY.
 - Trigger pulses are sent out correctly
 - Echo pulses *ARE NOT READ*

# B - Steps To Repeat

1. Connect the device up like so:
```
 ________ 
| S      |
| T      |          
| M      |          
| 3      |           ____________
| 2      |          |        H  |
| F   5V |----------| Vcc    C  |
| 3   PB0|----------| TRIG   -  |
| D   PB1|--[ 1K ]--| ECHO   S  |
| I   GND|----------| GND    R  |
| S      |          |        0  |
| C      |          |________4__|
| O      |
| V      |
| E      |
| R      |
|_Y______|
```
2. Launch the cleanflight GUI
3. Enable `SONAR` feature from 'Other Features' under Configuration tab
4. Navigate to 'Sensors' tab and check 'Sonar' option to see the distance measurements - always 0

# C - Video Links

N/A

# D - Forum post links

Discussed on Slack: https://cleanflight.slack.com/archives/C52RMJL01/p1515137572000149

# E - Version Information

Cleanflight: bd523ced7cc3a8c000068cca08da96eff43716a1
Configurator: v2.1.0

# E - Other information

# F - How to fix it!

The sonar drivers (`cleanflight/src/main/drivers/sonar_hcsr04.c`) use interrupts. The [`target.c`](cleanflight/src/main/target/STM32F3DISCOVERY/target.h) file only enables the interrupts if BMI160 is being used.

Recommend adding `#define USE_EXTI` following [line 192](https://github.com/cleanflight/cleanflight/blob/master/src/main/target/STM32F3DISCOVERY/target.h#L190-L192). Perhaps encapsulated in `#ifdef SONAR` condition.

mikeller, 2018-01-09:
@emmhaych:

> Recommend adding #define USE_EXTI following line 192. Perhaps encapsulated in #ifdef SONAR condition.

Can you please open a pull request for this?

hydra, 2018-03-07:
@mikeller has this been fixed in the BF codebase and recent merge?

hydra, 2020-05-17:
Please re-test in the latest code in `master` and if there's still an issue please re-open this.

closed 2020-05-17 by hydra