cleanflight/cleanflight #2997: Barometer (BMP180) dosn`t work on SP Racing F4 Neo

Kay01, 2018-01-07:
Hi together

Hopefully some of can help me. I try to install a Baro-Sensor (BMP180) on my copter with no success.

**FC:**    SP Racing F4 Neo with OSD Modul
**FW:**   CleanFlight V2.2.0 RC1 (stable)
**GPS:**    Neo7 (UART 3) incl. compass (I2C)
**Baro:**   Arduino BMP180 (I2C)

The baro sensor works fine on my Arduion Uno with 3.3 V. I use the sketch for the BMP085. The I2C-adress is 0x77. In the CleanFlight configurator (V 2.1.0) I enabled **Configuration** --> **Barometer** --> **On**

In the CLI I get the `status`:

> System Uptime: 591 seconds
> Voltage: 16 * 0.1V (0S battery - NOT PRESENT)
> CPU Clock=168MHz, GYRO=ICM20602, ACC=ICM20602, MAG=HMC5883
> SD card: None inserted
> Stack size: 2048, Stack address: 0x10010000
> I2C Errors: 0, config size: 2086, max available config: 16384
> CPU:5%, cycle time: 127, GYRO rate: 7874, RX rate: 49, System rate: 9
> Arming disable flags: RXLOSS CLI

There is no I2C Error.
My setup in the CLI is as following - `dump`:

> set mag_bustype = I2C
> set mag_i2c_device = 1
> set mag_i2c_address = 0
> set mag_spi_device = 0
> set mag_hardware = AUTO
> set mag_declination = 0
> set magzero_x = 0
> set magzero_y = 0
> set magzero_z = 0

(The Magnometer works fin and without any problems)

> set baro_bustype = I2C
> set baro_spi_device = 0
> set baro_i2c_device = 1
> set baro_i2c_address = 119 **_(Adress: 0x77)_**
> set baro_hardware = BMP085 **_(works fine on Arduino Uno)_**
> set baro_tab_size = 21
> set baro_noise_lpf = 600
> set baro_cf_vel = 985
> set baro_cf_alt = 965

I also checked the `resource`:

> resource I2C_SCL 1 B08
> resource I2C_SDA 1 B09

and the `resource list`:

> B08: I2C_SCL 1
> B09: I2C_SDA 1

All I have checked up to now I did it the Information of "google" and "Github". I am not a SW specialist. So it would be great you could help me.

THX, Kay

hydra, 2018-02-22:
Hi, currently the only compiled-in supported baros for the F4NEO are as follows:

```
#define USE_BARO_BMP280
#define USE_BARO_MS5611
```

Compile a custom version of CF and add the appropriate macro to the target.h file for the F4NEO.  Please report back and/or submit a PR to add it once you've done this and tested it.

hydra, 2018-03-30:
closing, no feedback.

closed 2018-03-30 by hydra