cleanflight/cleanflight #3007: SPRacing F3 NEO VTX not working

inagoch, 2018-02-06:
I am using a SPRacing F3 NEO stack, and the VTX seems not working. 
Can't change band/channel/power through OSD/CLI. 

I have tried cleanflight 2.0.5, 2.2, betaflight 3.1.7, 3.2.4.

hydra, 2018-02-22:
hi, please give more information about what isn't working for you.  like *EXACTLY* what isn't working and what *IS* working.  As it stands there's not enough information above to start diagnosing the problem.  

Certainly the code works on the boards I have here so it could be a hardware fault, configuration issue or wiring fault.

Please also reset firmware to defaults and re-test, do not restore a dump/diff/backup, etc.

hydra, 2018-03-07:
closing, no response.

closed 2018-03-07 by hydra