cleanflight/cleanflight #3008: Crazepony NRF communication issue

AsadMasood, 2018-02-07:
Hi,
I'm trying to make a swarm of crazepony quadcopters, I'm unable to communicate NRFL24L01+ from my PC to NRF on crazepony quadcopter? anyone help me out?
I wrote same code in C of NRF source code but from PC it is unable to communicate with the NRF of the crazepony quadcopter.

mikeller, 2018-02-08:
What hardware are you using? I could only find information on crazepony hardware that supports FrSky protocols, not NRF24.

AsadMasood, 2018-02-08:
Hardware is crazepony ,it has NRF24L01+

https://www.amazon.com/dp/B015MYR1GO

mikeller, 2018-02-08:
Looks like it's not supported by cleanflight or Betaflight. Ask the manufacturer to supply you with firmware and instructions for it.

closed 2018-02-08 by mikeller

AsadMasood, 2018-02-09:
I have asked but they won't reply.
Actually the issue I'm facing is in communication of NRF with the PC