cleanflight/cleanflight #3018: Please add feature to reduce Tricopter Tail-Wag.

ResinSmoker, 2018-02-20:
**# A - Description**

Would it be possible to add some sort of Servo "smoothing" so Tailwag wouldn't be so intense.
Often the Wag is out of control making the Tricopter either unflyable or near impossible to tune.
It's as if the Servo is constantly overshooting regardless of PID / CLI settings used.

**# B - Steps To Repeat**

Typically high frequency Wag is seen immediately after arming, before throttle up. After which the Wag dissipates while the throttle is being manually held above idle. Later low frequency Wag can often show up during sharp turns and fast straits. This makes the Yaw control sloppy at best, unflyable at its worst.

**# C - Video Links**

Examples:
Stationary wag https://www.youtube.com/watch?v=SeXPp4LuZms
Under Throttle https://www.youtube.com/watch?v=f6tYqGBGsXg

**# D - Forum post links**

There are tons of links as this is a very common issue for Tricopters.

**# E - Version Information**

Betaflight 10.1.0
Triflight on Betaflight 3.2.2 v2 / MATEKF405 (MKF4) 3.2.2 Dec 31 2017 / 15:24:22 (30f68c8) MSP API: 1.36

**# E - Other information**

-- Flight controller make/model : MATEK F405 AIO
-- Dump: https://pastebin.com/x3118UF8
-- Diff: https://pastebin.com/0YJesKyN
-- “Aikon” AK32 35A 6S BLHeli_32 ESC’s See image for more ESC details https://flic.kr/p/DXf5Kp
-- FrSky R-XSR with EU firmware from mfr (FW-RXSR-171103)
-- Taranis X9D (EU) FW: opentx-x9d+  Vers: 2.2.1  Date: 2017-12-17 EEPR: 218
-- Schematics (None)
-- Wiring diagrams: https://flic.kr/p/23sQEuz
-- Windows 10
-- Tricopter wired up with 470uf 35v capacitors for all ESC's and Battery.
-- Also have an F3FC based Tricopter from RCexplorer with the same issues.

mikeller, 2018-02-20:
@lkaino this might be of interest to you.

ResinSmoker, 2018-02-21:
Adding a Log file but unsure if it's here as I don't see it attached anywhere.

ResinSmoker, 2018-02-21:
Closing request and reposting to Betaflight

closed 2018-02-21 by ResinSmoker