cleanflight/cleanflight #3020: adding gps to debug mode

theSkonk, 2018-02-21:
would be nice to add gps data to the debug mode so an on board gps unit can logged ( personally for speed data )

ledvinap, 2018-02-21:
GPS logging is already implemented, as separate GPS frames, in blackbox ..

theSkonk, 2018-02-21:
really where ?

ledvinap, 2018-02-21:
https://github.com/betaflight/betaflight/blob/master/src/main/blackbox/blackbox.c#L235-L262
https://github.com/betaflight/betaflight/blob/master/src/main/blackbox/blackbox.c#L1471

theSkonk, 2018-02-21:
how does one access the data? i cant find any information on how to use it ( sorry if I'm a bit thick )

ledvinap, 2018-02-21:
Try https://github.com/cleanflight/blackbox-tools .. 
Not sure about blackbox-log-viewer, maybe it doesn't implement it. It would be quite easy to implement

theSkonk, 2018-02-21:
so put a request into the black box thread? its really not easy to use the tools there are no clear 'how to' information

ledvinap, 2018-02-21:
Did you read the README for blackbox tools?

theSkonk, 2018-02-21:
so I've worked it out.. its still really... really, REALLY complicated

hydra, 2018-02-22:
if you would like to help by improving things to make it easier please create documentation or code and/or submit a PR.  glad you worked it out though.   if you could explain which part in particular you had issues with that would help.

theSkonk, 2018-02-26:
hey man yeah the phrasing of the commands has to be exactly right as does the file your using and if your not quite familiar with running command line stuff the read me file doesn't give an example so the less smart know what to do..
if the info is embedded in the black box logs would it be east to added it as a plot-able in black box?

hydra, 2018-03-07:
this is going off-topic now.  closing.  if you have ideas, etc then please create a new issue.  for help using the software please use forums or slack.

closed 2018-03-07 by hydra