cleanflight/cleanflight #3023: Cannot run program make/system-id.mk:49: *** failed to detect operating system.

Jasonsiu, 2018-02-24:
Hi, guys.
I am trying to build the cleanflight code in Eclipse by following the steps in the link
https://github.com/cleanflight/cleanflight/blob/master/docs/development/Building%20in%20Eclipse.md
and i am not able to build the program after importing the same.
I am getting the error as mentioned below,

Cannot run program "failed to detect operating system. "

make TARGET=CC3D DEBUG=GDB 
uname reports 
make/system-id.mk:49: *** failed to detect operating system.  Stop.
uname -m reports 

Kindly help me out in this.

My operating system is Windows 10( 64 bit)
Eclipse IDE for C/C++ Developers Version: Oxygen.2 Release (4.7.2) (64 bit)
CYGWIN64 
GNU MCU Eclipse Build Tools 2.10-20180103-1919

mikeller, 2018-02-24:
Hi @Jasonsiu. Have you tried building the firmware from the command line first? If not, please do. Making sure that your build environment is working **before** trying to build from within an IDE is crucial, as the IDE can hide errors, or cause others.

Also, you might want to look into this: https://github.com/betaflight/betaflight/wiki/Building-Betaflight-on-Windows,-the-Linux-subsystem-way. This seems to be an easier way to build under Windows, compared to cygwin.

hydra, 2018-03-07:
closing, no response.

closed 2018-03-07 by hydra