cleanflight/cleanflight #3034: SP Racing F3 isn't moving any servos as COSTUMAIRPLANE ,But have neutral position signal output

bluecat1997, 2018-03-24:
# A - Description

-- I have a SPRacing F3 Flight Controller . And for a project I need to mix two motor with throttle and pitch . And mix Servo with ROLL and YAW. 

But after I set the mixer and remapped Pin, My logic analyzer shows that MOTOR 1 and MOTOR 2 works well. BUT the SERVO 1 and SERVO 2 only shows correct by the Servos bar of Motors tag in the CleanFlight software. 
[image]

But only Output  neutral position signal at the Pin I choose in Remapping .
[image]

    **Here is the mixer:**
`# mixer
Mixer: CUSTOMAIRPLANE

# mmix
mmix 0  1.000  0.000 -1.000  0.000
mmix 1  1.000  0.000  1.000  0.000

# smix
smix 0 0 0 100 0 0 100 0
smix 1 0 2 -100 0 0 100 0
smix 2 1 0 100 0 0 100 0
smix 3 1 2 100 0 0 100 0`

**And Remapping like this :**
`resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 7 A02
resource MOTOR 8 A03
resource SERVO 1 B08 #B08 Is for MOTOR 5 before
resource SERVO 2 B09 #B09 Is for MOTOR 5 before`

**SERVO Command in CLI is below (By the way,I can't understand the syntax of command "SERVO", especially the last -1)** 
`servo 0 1000 2000 1500 100 -1
servo 1 1000 2000 1500 100 -1
servo 2 1000 2000 1500 100 -1
servo 3 1000 2000 1500 100 -1
servo 4 1000 2000 1500 100 -1
servo 5 1000 2000 1500 100 -1
servo 6 1000 2000 1500 100 -1
servo 7 1000 2000 1500 100 -1` 

# B - Steps To Repeat
null

# C - Video Links
null

# D - Forum post links
null

# E - Version Information

# version
# Cleanflight / SPRACINGF3 (SRF3) 2.2.0 Mar  3 2018 / 13:34:38 (577ab43) MSP API: 1.37

Configurator: v2.1.0

# E - Other information
null

mikeller, 2018-03-25:
What does `resource list` tell you?

hydra, 2018-09-28:
this issue is this bit of code:

```
    case MIXER_CUSTOM_AIRPLANE:
    case MIXER_AIRPLANE:
        for (int i = SERVO_PLANE_INDEX_MIN; i <= SERVO_PLANE_INDEX_MAX; i++) {
            pwmWriteServo(servoIndex++, servo[i]);
        }
        break;
```
the first servo that is output is SERVO_FLAPS which is 2.

try this instead:

```
smix 0 2 0 100 0 0 100 0
smix 1 2 2 -100 0 0 100 0
smix 2 3 0 100 0 0 100 0
smix 3 3 2 100 0 0 100 0
```

i tested it just now and it works for me.

closed 2018-09-28 by hydra