cleanflight/cleanflight #3035: Mismatch timers and motors

commonnick, 2018-03-28:
Hello! I have a F4-based custom board with 6 brushed motors.
I've created target.c with timers and motors defined like following: 

```
const timerHardware_t timerHardware[USABLE_TIMER_CHANNEL_COUNT] = {
DEF_TIM(TIM1, CH3, PE13, TIM_USE_MOTOR, TIMER_OUTPUT_ENABLED, 0),
DEF_TIM(TIM1, CH4, PE14, TIM_USE_MOTOR, TIMER_OUTPUT_ENABLED, 0), 
DEF_TIM(TIM2, CH4, PB3, TIM_USE_MOTOR, TIMER_OUTPUT_ENABLED, 0),
DEF_TIM(TIM3, CH1, PA6, TIM_USE_MOTOR, TIMER_OUTPUT_ENABLED, 0),
DEF_TIM(TIM4, CH2, PD13, TIM_USE_MOTOR, TIMER_OUTPUT_ENABLED, 0),
DEF_TIM(TIM9, CH1, PE5, TIM_USE_MOTOR, TIMER_OUTPUT_ENABLED, 0),
};
```

I observe that in mixer.c:102 
motorConfig->dev.ioTags[motorIndex] it filled in correctly. 

But in pwm_output.c:282
const timerHardware_t *timerHardware = timerGetByTag(tag, TIM_USE_ANY) returned NULL. 
What is the correct way to configure brushed motors ? Do I have to replace TIM_USE_ANY to TIM_USE_MOTOR in pwm_output.c? 
Thanks

mikeller, 2018-03-28:
No, that should all work fine, resources for brushed motors and brushless motors are defined in the same way. What does `resource list` in CLI tell you?

commonnick, 2018-03-28:
```
# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: FREE
A05: FREE
A06: MOTOR 4
A07: FREE
A08: I2C_SCL 3
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: FREE
A12: FREE
A13: FREE
A14: FREE
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: MOTOR 3
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: MPU_EXTI
B09: COMPASS_EXTI
B10: FREE
B11: FREE
B12: FREE
B13: FREE
B14: FREE
B15: FREE
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: I2C_SDA 3
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: MOTOR 5
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: MOTOR 6
E06: FREE
E07: FREE
E08: LED 1
E09: FREE
E10: LED 2
E11: FREE
E12: FREE
E13: MOTOR 1
E14: MOTOR 2
E15: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: SERIAL_TX 1
```

mikeller, 2018-03-28:
This looks all good - none of the 6 motors have a resource conflict. Without the target definition it is hard to tell what is going wrong.

closed 2018-03-30 by hydra