cleanflight/cleanflight #3038: first drone: "failed to connect to serial port" - NEED HELP :)

Driyyel, 2018-03-30:
hi guys,
I'm building my first drone at the moment, and am stuck at the configuration process of my SP racing F3 flight controller. I installed all the drivers that are recommended at the cleanflight welcome menu, and tried to fix my issue with the impulseRC driver fixer... didn't work. also did some research, but none of the solutions seemed to function for me. here's a picture with the error, if that helps in any way, the fc was connected when I took the screenshot
[image]

If anyone knows how to fix this issue, please tell me, I'm lost
thanks,
timo

mikeller, 2018-03-30:
Have you tried with a different USB cable?

Driyyel, 2018-03-30:
sadly didn't make any difference... but thanks anyways. any other idea?

hydra, 2018-03-30:
please can you use a support forum or slack for support rather than github. thanks.

closed 2018-03-30 by hydra