cleanflight/cleanflight #3044: Issues with the OSD on the SPracingF3NEO & SPracingF3EVO OSD/PDB with version

heliboy, 2018-04-26:
# A - Description
When updating SPRACINGF3NEO or the SPRACINGF3EVO the OSD stops working, providing awaiting for Flight controller in the video link.  Further more when clicking the OSD tab in the configurator it hangs the configurator, requiring it to be closed and reopened.  Even after reseting the board or the configurator the fault persists...  Reflashing with 2.2.0 fixes the issue.

# B - Steps To Repeat
Reflash with 2.3.

# C - Video Links
Will provide a video tomorrow.

# D - Forum post links
none

# E - Version Information
2018-04-26 @ 22:39:32 -- Running - OS: MacOS, Chrome: 65.0.3325.181, Configurator: 2.1.0
2018-04-26 @ 22:46:50 -- Using cached release information.
2018-04-26 @ 22:46:56 -- Serial port successfully opened with ID: 8
2018-04-26 @ 22:46:56 -- MultiWii API version: 1.38.0
2018-04-26 @ 22:46:56 -- Flight controller info, identifier: CLFL, version: 2.3.2
2018-04-26 @ 22:46:56 -- Running firmware released on: Apr 23 2018 00:41:09
2018-04-26 @ 22:46:56 -- Board: SP3N, version: 0
2018-04-26 @ 22:46:56 -- Unique device ID: 0x3700335633570e20333131

# E - Other information
-- flight controller SPRACINGF3NEO v0
-- esc make/model/config - Wraith 
-- receiver make/model/config FRSKY XSR SBUS
-- transmitter make/model/config Traranis X9D PLUS
-- operating system - MacOS High Sierra 10.13.4
-- etc.

-- config dump + diff
# DIFF
# Cleanflight / SPRACINGF3NEO (SP3N) 2.3.2 Apr 23 2018 / 00:41:09 (208d4cc) MSP API: 1.38
[image]
[image]
profile 0
rateprofile 0

# DUMP
# Cleanflight / SPRACINGF3NEO (SP3N) 2.3.2 Apr 23 2018 / 00:41:09 (208d4cc) MSP API: 1.38
name -
resource BEEPER 1 C15
resource MOTOR 1 B01
resource MOTOR 2 C07
resource MOTOR 3 B00
resource MOTOR 4 C06
resource MOTOR 5 B10
resource MOTOR 6 B11
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A02
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
[... 534 lines cut ...]
set p_yaw = 70
set i_yaw = 45
set d_yaw = 20
set p_level = 50
set i_level = 50
set d_level = 75
set level_limit = 55
set horizon_tilt_effect = 75
set horizon_tilt_expert_mode = OFF
rateprofile 0

set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

benderl, 2018-05-14:
This firmware version requires an updated configurator. I had the same issue with version 2.1.0 of the configurator. Please read the release notes. You will need version >= 2.2.0 RC1. Unfortunately this version is not available in the chrome webstore. You have to install it by source.

danielegger1, 2018-06-13:
I have the same Issue with the SPRacingF4Evo combined with the SPRacingF3OSD.

hydra, 2018-09-28:
yes, known issue, see release notes for latest firmware. closing as fixed.

closed 2018-09-28 by hydra