cleanflight/cleanflight #3045: spracing f7dual , esc telemetry and camera control no pins defined

emski0, 2018-04-27:
Hi
I am trying to have the esc telemetry working, but looking into the code ,seems the pin and timers are not defined.
can anybody guide me to add the code line to the board target, and i compile it myself.
I want to use the esc telemetry on UART 5 RX,(which is shared withe bind button) i don't need the bind botton. 
i see these line in the code : 
// TODO
 //#define USE_ESCSERIAL
// #define ESCSERIAL_TIMER_TX_HARDWARE 0 // PWM 1

i see other board using #define ESCSERIAL_TIMER_TX_PIN ,
also it needs timer in target.c : 
// For ESC serial
    DEF_TIM(TIM9, CH1, PA2,  TIM_USE_NONE,  0, 0 ), 
Can anybody help me, i don't mind using the timer of the transponder or led, since i am not using them.
Thanks

mikeller, 2018-05-04:
You are barking up the wrong tree here, @emski0. The ESCSERIAL functionality is what is used by the `escprog` serial port to ESC forwarding functionality, and **not** by ESC telemetry. ESC telemetry can be used on any UART RX pin (hard or soft serial).

emski0, 2018-05-04:
@mikeller i didn't know , i thought it's for the esc sensor,
I modified the first post and title

hydra, 2018-09-28:
yes you are correct, the bind buttons is on the same pin as uart5_rx.  Yes, ideally the bind button code needs to be able to be disabled in software.  There is work going on to enable this kind of thing in betaflight 4.0, for now doing what you suggest and building custom firmware is the solution for you.

hydra, 2018-09-28:
Should also note that VTX control button ideally also needs to be configurable independently from the bind button.  The button code also needs to be re-worked so that it doesn't pause the main loop waiting for the button to be released, but doing that without allowing it to be disabled is fairly pointless since any device connected to UART5_RX would likely cause the FC to think someone was pressing the button.

hydra, 2020-05-17:
I think, with the code in `master` now you should be able to use the `cli` resource command to remove the RX pin from UART5.  If this is still an issue and that doesn't work, please re-open this issue.

closed 2020-05-17 by hydra