cleanflight/cleanflight #3046: building and flashing cleanflight using eclipse

amitmanish, 2018-04-29:
Hi guys,
I wanted to build and flash cleanflight on flight controller using eclipse. I have seen build steps on eclipse but could not find flashing steps. This is for learning the cleanflight codebase. I have naze32 board. How i can achieve this.

mikeller, 2018-05-03:
You need to use the cleanflight configurator to flash the firmware: https://github.com/cleanflight/cleanflight-configurator

Scavanger, 2018-05-10:
Its possible but you need ST-Link V2 or another debug hardware (see https://github.com/cleanflight/cleanflight/blob/master/docs/development/Hardware%20Debugging.md).  Its not possible over the onboard Serial USB port.
But be warned: Config Eclipse for Cleanflight debugging can cause some pain. When everything is correct installed, just hit the debug button in eclipse and the is compiled and flashed on the board in one step.
Naze32 isn't supported anymore and i think the Naze32 has no connections for debug hardware.  Use F3 or F4 (better) boards for development.

Scavanger, 2018-07-22:
Sure, you can easily add stm32flash (https://sourceforge.net/projects/stm32flash) as an postbuild step.
But what you really want ist hardware debugging, believe me. ;)

hydra, 2018-09-28:
closing as stale.

closed 2018-09-28 by hydra