cleanflight/cleanflight #3047: Telemetry not working on SP Racing F7 Dual

tonipalmgren, 2018-04-29:
FlySky IBUS and SmartPort telemetry do not work on an SP Racing F7 Dual. I'm using the dedicated receiver port; serial on R2, SmartPort on T5, and I've also tried SmartPort on UART4 and UART1.

I would had tried softserial, but it isn't available; it's always disabled after rebooting.

I'm using a FlySky FS-I6X transmitter with an FS-ia6b receiver, and a FrSky Taranis X9D+ with an X8R receiver.

Alofly, 2018-04-30:
Hi, I have the same trouble with de FrSky R-XSR on CleanFlight 2.2.0 and 2.3.2

dydek, 2018-05-01:
I was trying to set up it too - using UART2 with IBUS telemetry configured without any luck ( cleanflight  2.3.2 , FlySky iA6b receiver and FS-I6 transmitter )

tonipalmgren, 2018-05-01:
Ok. Good to know. I've updated every firmware and tried everything that has come to my mind. Unfortunately I've already ordered another SP Racing F7 Dual and it's too late to cancel. I hope this issue will be resolved quickly. I've got one build and another repair on hold until this gets resolved, as it's total insanity to fly without a low voltage warning.

micworg, 2018-05-13:
Same issue here .. hopefully this can be fixed soon.

tonipalmgren, 2018-05-14:
@micworg What hardware were you using? Flight controller, receiver, and transmitter?

hydra, 2018-05-19:
Seems rather odd, I'll check it soon - I fly FrSky with XSR's most of the time and telemetry has always worked for me on the SPRacingF7DUAL.

hydra, 2018-05-19:
confirmed X8R + Taranis Open Tx 2.1.7 works with SBUS on UART2, SmartPort on UART5 with the following setting diff:
```
# Cleanflight / SPRACINGF7DUAL (SP7D) 2.3.2 May 15 2018 / 15:54:03 (208d4cce4) MSP API: 1.38
set tlm_inverted = OFF
```

Alofly, 2018-05-19:
I have configured my F7 in the same way I did in my F3NEO and F4EVO. 
But on the F7 it does not work. What would I have to do differently in F7 to make it work?

If there really is nothing different to do ... What would I have to do to make you believe what I say?

tonipalmgren, 2018-05-19:
@hydra "tlm_inverted = off" seems to have "fixed" it. I hadn't found this command, but hey, I only spent hours and hours searching on Google and watching videos. :P I saw you create a Wiki-type page for the S7 Dual. Could that be a nice place for this kind of info? A place to keep all the F7 Dual related stuff in, regardless of other hardware used. Would had saved a lot of headache... I was also under the impression that turning on SmartPort on some port would automatically turn invertion off, if needed, which is partly why I stopped googling after only a gazillion hours... :P

So, to sum it up, "set tlm_inverted = off" is needed with Taranis X9D+ and X8R to get the telemetry to work. I'm moving from FlySky to FrSky, so I won't test FlySky telemetry for now, but maybe later.

@Alofly All I did was "set tlm_inverted = off" in the Cleanflight CLI, then typed "save" and everything worked. Though I have the X8R, not the XSR.

micworg, 2018-05-19:
"tlm_inverted = off" fixed it for me too.

tonipalmgren, 2018-05-19:
@Alofly I suggest you try the "set tlm_inverted = off" in the command line. This seems to work, at least for the FrSky products.

tonipalmgren, 2018-05-19:
"set tlm_inverted = off" fixes this on the FrSky products.

closed 2018-05-19 by tonipalmgren

Alofly, 2018-05-19:
tonipalmgren, you're right. I tried it before but I did not put the "save" command at the end

Thank you so much for your patience