cleanflight/cleanflight #3049: SeriouslyPro F7 board - No Motor outputs on motors 5 to 8

zedthirteen, 2018-05-07:
# A - Description
Motor outputs 5 to 8 not working on SPF7.  I have configured my rig as an HEX-X machine. Motors 1 to 4 work fine but motors 5 and 6 do not work.

I have also tried configuring my rig as an octo and moving motors 5 & 6 to outputs 7 & 8 but this did not work either.

ESCs and motors 5 & 6 have been proven to be working by connecting temporarily to outputs 3 and 4.

# B - Steps To Repeat

Connect rig to CF configurator and apply battery power.  Switch to motors page. Enable motors and then increase master slider.  The CF motor power indicators for 1 to 6 show the selected power but only motors 1 to 4 spin.

# C - Video Links
Short video here: https://youtu.be/niTKH9E-rRU

photos of wiring and configuration here:
http://s1195.photobucket.com/user/zed_thirteen/library/Seriaously%20Pro%20F7%20Configuration%20Problems
I know one of those motor connections looks poorly attached but it is solid and all motor wires have continuity to the ESC connector. There are no shorts to adjacent pins, ground or 5v.

Note the outputs on the logic analyser image.  Motor 5 is permanently high and motor 6 is permanently low

# D - Forum post links

-- provide links to forum posts if you have discussed this with others --

# E - Version Information

Cleanflight: v2.3.1 and 2.3.2 tried
Configurator: v2.2.0

# E - Other information
-- flight controller make/model = SPF7 and stacking VTX with TBS Unify Pro SmartAudio
-- config dump attached 
[CLFL_cli_20180507_101742.txt](https://github.com/cleanflight/cleanflight/files/1979486/CLFL_cli_20180507_101742.txt)

-- ESCs are all ONESHOT125
-- receiver make/model/config = Orange Spektrum Satellite
-- transmitter make/model/config = Spektrum DX8

hydra, 2018-05-08:
It's likely the issue is caused by the removal of the FEATURE_VTX in betaflight changes since the board was designed.  It's trying to use the M5-M8 pads for VTX.

I don't think this showed up recently because I enabled the M5-M8 motor outputs in the code first, did all the testing, and then later added the VTX support code knowing, so I thought, that the M5-M8 pins would be unused by the VTX code untill I enabled the `VTX` feature, however the VTX feature had already been removed by the time the SPRacingF7DUAL code changes were merged.

hydra, 2018-05-08:
To confirm this, here's a version of CF 2.3.2 without the VTX6705 code compiled in.

@zedthirteen Please can you test this file and report back.

[cleanflight_2.3.2_SPRACINGF7DUAL-NO_VTX.zip](https://github.com/cleanflight/cleanflight/files/1982899/cleanflight_2.3.2_SPRACINGF7DUAL-NO_VTX.zip)

Here's the diff of the code changes.

```
diff --git a/src/main/target/SPRACINGF7DUAL/target.h b/src/main/target/SPRACINGF7DUAL/target.h
index a0855f432..f99f2bf85 100644
--- a/src/main/target/SPRACINGF7DUAL/target.h
+++ b/src/main/target/SPRACINGF7DUAL/target.h
@@ -158,14 +158,14 @@
 #define SDCARD_DMA_CHANNEL_TX_COMPLETE_FLAG DMA_FLAG_TCIF7_4
 #define SDCARD_DMA_CHANNEL                  DMA_CHANNEL_0

-#define USE_VTX_RTC6705
-#define USE_VTX_RTC6705_SOFTSPI
-#define VTX_RTC6705_OPTIONAL    // VTX/OSD board is OPTIONAL
-
-#define RTC6705_SPI_MOSI_PIN                PB0  // Shared with PWM8
-#define RTC6705_CS_PIN                      PB6  // Shared with PWM5
-#define RTC6705_SPICLK_PIN                  PB1  // Shared with PWM7
-#define RTC6705_POWER_PIN                   PB7  // Shared with PWM6
+//#define USE_VTX_RTC6705
+//#define USE_VTX_RTC6705_SOFTSPI
+//#define VTX_RTC6705_OPTIONAL    // VTX/OSD board is OPTIONAL
+//
+//#define RTC6705_SPI_MOSI_PIN                PB0  // Shared with PWM8
+//#define RTC6705_CS_PIN                      PB6  // Shared with PWM5
+//#define RTC6705_SPICLK_PIN                  PB1  // Shared with PWM7
+//#define RTC6705_POWER_PIN                   PB7  // Shared with PWM6

```

hydra, 2018-05-08:
Confirmed using the scope, with latest official firmware outputs M5-M8 are incorrectly used by the vtx code.  Using the firmware on my previous post I confirm that M5-M8 provide motor outputs as expected.

zedthirteen, 2018-05-08:
Wow, that is a very fast response. I can't test it for a few days but I will let you know how I get on
Cheers

zedthirteen, 2018-05-08:
I can confirm that I can now control all 6 motors through CF configurator
Thanks

hydra, 2018-07-16:
Here's an updated binary for SPRacingF7DUAL (no VTX version, as above) based on CL 2.4.0-RC1.

Only for users needing all motor outputs 1-8.

[cleanflight_2.4.0_SPRACINGF7DUAL-NO_VTX.zip](https://github.com/cleanflight/cleanflight/files/2199078/cleanflight_2.4.0_SPRACINGF7DUAL-NO_VTX.zip)

hydra, 2018-10-01:
Update binary for SPRacingF7DUAL (no VTX version, as above) but based on CLFL 2.5.0-RC1.

Again, only for users need all motor outputs 1-8.

[cleanflight_2.5.0_SPRACINGF7DUAL-NO_VTX.zip](https://github.com/cleanflight/cleanflight/files/2432507/cleanflight_2.5.0_SPRACINGF7DUAL-NO_VTX.zip)

zedthirteen, 2018-10-26:
Hi Dominic,

 

Sorry about my previous message. I didn’t mean to contact you from my spam mail account! Ntot sure if you got it either as appears to have been a reply to github message

 

While looking at CLI settings I started to tinker with GYRO_TO_USE and tried FIRST and SECOND.  These both took care of my gyro X & Y sensors moving together but neither corrected my problem.

 

Are there known working settings for the F7Dual in hex-x?

 

Regards

David

(Hopefully from my normal email account)

 

From: Spam Account [mailto:spam@davidandemma.com] 
Sent: Friday, October 26, 2018 12:23 PM
To: 'cleanflight/cleanflight'
Subject: RE: [cleanflight/cleanflight] SeriouslyPro F7 board - No Motor outputs on motors 5 to 8 (#3049)

 

Hi Dominic,

 

Sorry to bother you with this but I haven’t been able to get my SP F7 Dual board into the air.

 

I have tried the Facebook user group but not had any responses.

 

I’m running a Hex-X Tarot Iron man frame with the latest NO-VTX config.

 

https://github.com/cleanflight/cleanflight/files/2432507/cleanflight_2.5.0_SPRACINGF7DUAL-NO_VTX.zip

 

I have removed the VTX board, GPS, MAG, OSD and LEDs to try to minimise what might be causing my issue. 

 

F7 board is installed with SD card up and USB socket to the right

 

I have checked all motors are working and spinning in the correct direction. In the config screen the simulated Hex-X responds in the same way as my frame. 

 

In the sensors screen my accelerometer Z axis is reporting 1G and the X and Y axis report 0 but  respond correctly if I move the machine forwards/backwards and left/right respectively. With machine pointing upwards then X = 1. With machine right motor pointing upwards then Y = 1. These values seem correct to me.

 

If I rock the machine front to back then gyro X&Y appear to follow an identical sine wave and Z stays pretty stable.  If I rock from right to left then the X&Y sine waves are in opposition to each other.  When I rotate the machine back and forth around the Z axis then the Z trace describes a sine wave.  It’s probably normal but I don’t understand why the X&Y traces act together when Z appears to act on its own – could this be causing my problem?

 

When I arm my machine the motors start spinning and if give commands for pitch or roll the machine looks like it is applying thrust to the correct motors

 

i.e. 

stick left = right hand motor 5 spins up

stick right = motor 6 spins up

stick forward = motors 3 & 1 spin up

stick back = motors 4 & 2 spin up

 

as I apply throttle to take of the machine appears to want to spin or flip with props hitting the dirt.

 

Any help will be appreciated.  I’ve attached a CLI dump in case it helps

 

On connection to Cleanflight my board reports:

2018-10-26 @ 12:14:10 -- Board: SP7D, version: 0

2018-10-26 @ 12:14:10 -- Unique device ID: 0x4200473037510f35333131

 

I saw mention of a rev 1 board in one of the git comments online – is there a difference?

 

Sorry for the long email but I wanted to give you any relevant information that I could think of.

 

How is Portugal?

 

Regards

David

 

 

 <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> 

Virus-free.  <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> www.avg.com

hydra, 2018-11-21:
Please note, users needing to use M5/M6 outputs currently will either need to disable DSHOT BURST (aka DSHOT DMAR) or disable LED STRIP feature.  This is due to a hardware resource conflict between DSHOT BURST and LED STRIP.

DSHOT BURST is a software feature that was added /after/ the release of the SPRacingF7DUAL.  Using normal DSHOT is fine, DSHOT BURST uses a little less CPU time which is beneficial on older slower processors like the F3/F4.

hydra, 2020-05-17:
Current solution for this is to use the cli `resource` command to un-assign the pins used by the VTX.

closed 2020-05-17 by hydra