cleanflight/cleanflight #3051: SeriouslyPro F7 board - OSD Font upload never finishes

zedthirteen, 2018-05-13:
# A - Description

After requesting a new Font to be uploaded, CleanFlight configurator changes to "Uploading" but never finishes the upload

# B - Steps To Repeat

Connect to F7 board through CleanFlight configurator.  Change to the OSD page. Select Font to upload and click the "Upload" button.  The status changes to "Uploading" but never completes

# C - Video Links

N/A

# D - Forum post links

N/A

# E - Version Information

Cleanflight: v2.3.2 (build with motor output fix)
Configurator: v2.2.0

# E - Other information
SPF7 board with VTX module and TBS Unify Pro SmartAudio
Hex-X with OneShot125 ESCs
Orange Spektrum Satellite
Spektrum DX8
Windows 10 Pro 64Bit
Intel i5-4750K 16GB RAM

zedthirteen, 2018-06-07:
Found the F7 version 2 manual today.  I have now made the three links on VTX board and enabled "TBS smart audio" on UART 4 but the font upload still does not show any progress or finish - just stuck on "Uploading..."

hydra, 2018-08-07:
is this still an issue?  Not had any problems here.

hydra, 2018-08-07:
Just tested OK using:
Configurator 2.4.0
CF 2.4.0-RC1

No issues, closing.  Please create new issue if this is still an issue for you.

closed 2018-08-07 by hydra