cleanflight/cleanflight #3052: Msp

johthu, 2018-05-13:
Hi,

Could we get msp telem back?

https://github.com/cleanflight/cleanflight/issues/1278
This says it's msp but apparently this has been removed.

And how would you tap into this to receive sensor data from the board like the IMU and stick positions and transmit data like PID values etc. Since the configurator can do all this how can I?

Thanks alot!

mikeller, 2018-05-14:
@johthu: Have you tried `MSP_RC`, `MSP_ATTITUDE`, `MSP_PID` (and many more)?

johthu, 2018-05-14:
Hi, thanks for the reply,
Do you mean use those commands in the cli?

johthu, 2018-05-14:
Because they don't seem to work

mikeller, 2018-05-14:
No, they are MSP commands that you can use to get sensor data, IMU data, and stick positions over MSP.

johthu, 2018-05-15:
How do I use these?

mikeller, 2018-05-15:
The same way you'd use MSP telemetry.

Have a look at https://github.com/cleanflight/cleanflight-configurator/blob/master/src/js/msp/MSPHelper.js for application examples.

johthu, 2018-05-17:
Ah thanks, would it be possible to write this in c or python?

mikeller, 2018-05-18:
Yes, sure is.

mikeller, 2018-05-19:
@johthu: Have you looked into using one of the libraries that are out there that implement an MSP client, like https://github.com/fdivitto/MSP ?

hydra, 2018-09-28:
closing as stale.

closed 2018-09-28 by hydra