cleanflight/cleanflight #3057: OSD feature won't enable

bhomma, 2018-06-16:
# 3. If you answered NO to the first question.  Delete everything above this line, read the guidelines (see link above this form) and continue.

# A - Description

The osd feature in the cleanflight configurator will not stay enabled after I set the osd feature to "on" and then click "save and reboot"

# B - Steps To Repeat

FC is flashed with latest naze firmware.  Connect to configurator and go to the configuration tab in cleanflight.  Go to the features menu and enable osd feature.  Click save and reboot.  

# C - Video Links

-- provide a link to a video showing your problem, as applicable --

# D - Forum post links

-- provide links to forum posts if you have discussed this with others --

# E - Version Information

Cleanflight: vX.X.X or commit SHA
Configurator: vX.X.X or commit SHA

# E - Other information

-- provide additional information such as.
-- flight controller make/model
-- config dump + diff
-- esc make/model/config
-- receiver make/model/config
-- transmitter make/model/config
-- schematics
-- diagrams
-- operating system
-- etc.

mikeller, 2018-06-16:
NAZE does not have OSD hardware.

closed 2018-06-16 by mikeller

bhomma, 2018-06-16:
the board is the lluminati 32 with built in osd

bhomma, 2018-06-16:
it says in the configurator that some feature combinations are considered invalid is it possible that I have some features enabled that would cause it to automatically disable the osd feature?

mikeller, 2018-06-17:
If you'd provided the information that you were asked to provide when opening this issue then you probably could have been helped the first time round...

This board is not supported by cleanflight. Ask the manufacturer / supplier to provide you with updated firmware.