cleanflight/cleanflight #3063: Building in CLion?

TwoToneEddy, 2018-08-08:
I saw an issue from a couple of years back about building in CLion. Does anyone do this? I'm fairly familiar with CLion but i'm not sure how to configure the toolchain to use the relevant gcc-arm compiler.

mikeller, 2018-08-09:
@TwoToneEddy: I am not using CLion, or too familiar with it. But in the case of a lot of other IDEs (eclipse, Visual Studio Code), the commonly used approach is to use the IDE for its platform independent benefits (intellisense, ...), but then do the actual building with what is called a 'Makefile build' or suchlike, that under the hood just calls make with the appropriate options.

TwoToneEddy, 2018-08-09:
@mikeller Yeh I was thinking I'll probably just do that. How would you go about invoking all the defines in the target.h file of the given target in the IDE so all the correct header files are included? It would be good to have all the correct highlighting and searching the the idea but for example, as I haven't specified what chip it is (STM32F3 for example) there's allot of red. I assume this is usually done during the make.

mikeller, 2018-08-09:
@TwoToneEddy: Yes, you are correct there - I have not yet found an IDE that would be good enough at analysing the Makefile to come up with an include structure to make this work out of the box, without manual setup.
In broad strokes, what you want to do is to define the following includes:
- `lib/main`;
- `src/main` with all the files, but none of the directories under `src/main/target`;
- the one directory under `src/main/target` for the target you want the defines to apply for.

This should get you pretty much there. If any of the defines (like the `STM32F<x>` for the MCU type that gets defined on the fly in the Makefile) is still missing, define it manually in the IDE.

hydra, 2018-09-28:
closing as stale, no response.  thanks for the suggestions @mikeller

closed 2018-09-28 by hydra