cleanflight/cleanflight #3076: BARO mode not available (SP Racing F3 deluxe)

tomasssk, 2019-01-22:
Using the latest 2.5.0 Cleanflight firmware and SPRACINGF3 deluxe board, I want to be able to enable the ALT hold (BARO) mode. But it is not listed in MODES tab.
However when I go to SENSORS tab, the barometer values are logged correctly...
Do I need to enable it via the CLI command? Or how do I do it?

mikeller, 2019-01-22:
@tomasssk: BARO (altitude hold) had not been working properly for some time, and has been removed from recent versions of the firmware.

tomasssk, 2019-01-23:
Ok, what are my options then if I want the altitude hold? Betaflight?

mikeller, 2019-01-23:
@tomasssk: Same deal in Betaflight (cleanflight is 98% Betaflight). Try iNav (https://github.com/iNavFlight/inav/wiki).

closed 2020-05-17 by hydra