cleanflight/cleanflight #3078: Can not select flight modes with GPS in cleanflight 2.5

rodolfot, 2019-01-30:
Hello!

I have an SP Racing F3 with a custom version of Cleanflight (2.5+ GPS).
My controller is recognizing the GPS M8N perfectly, however, it is not possible to configure the flight modes using GPS.
I tried to add flight modes via code but I did not succeed. Is there any way to add GPS flight modes by code or command line (CLI)?

Thank you!

mikeller, 2019-01-30:
@rodolfot: You should be able to select GPS rescue, which is cleanflight's GPS related flight mode.

rodolfot, 2019-01-30:
It certainly must be without the GPS rescue. How can I check this? Thankful!
Another question.
If the GPS Rescue Mode does not appear, does the GPS HOLD mode also not appear? I ask this, because I wanted to use this flight controller with GPS.

Thank you very much

rodolfot, 2019-01-30:
DUMP:
[CLFL_cli_20190130_195938_dump.txt](https://github.com/cleanflight/cleanflight/files/2814641/CLFL_cli_20190130_195938_dump.txt)

HEX:
[cleanflight_2.5.0_SPRACINGF3.hex.zip](https://github.com/cleanflight/cleanflight/files/2814627/cleanflight_2.5.0_SPRACINGF3.hex.zip)

Configuration images:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

mikeller, 2019-01-31:
@rodolfot: Your chances are better to get your issue fixed in iNav - cleanflight has not had a focus on autonomous modes in a long time, and the code for it was unmaintained and not working properly for some time before it was finally removed.