cleanflight/cleanflight #3080: SPRacingF3 can not connect to configurator with cygwin built hex file

ygntae, 2019-02-10:
Hello,

When ı flash the stabil hex from git to SPRACINGF3 board, everything work well. But when ı flash hex file created by cygwing, the configurator gives the following error;

2019-02-10 @ 20:27:43 -- Serial port successfully opened with ID: 24

2019-02-10 @ 20:27:45 -- MultiWii API version: 1.40.0

2019-02-10 @ 20:27:53 -- No configuration received within 10 seconds, communication failed

2019-02-10 @ 20:27:53 -- Serial port successfully closed

mikeller, 2019-02-10:
@ygntae: Looks like your build is not working. Are you building from unchanged sources, or do you have local modifications?

ygntae, 2019-02-11:
@mikeller hi,

I had no modification. The interesting thing is thah, ı tried to connect betaflight configurator with cleanflight 2.4.1 built. It could only detectd CLI mode and when ı ask version, the board sends version information to the betaflight configurator CLI screen. So the other interesting thing is, with the 2.4.1 version, it seems like the board is about to connect to configurator but the 10 seconds does not enough for it while connectiong to cleanflight configurator.

mikeller, 2019-02-11:
@ygntae: Seeing the seemingly erratic behaviour that you describe, I suspect that the cause might be that the USB cable you are using is not making a proper connection. Try with a different USB cable.

hydra, 2020-05-17:
please retest with latest code from `master` and re-open if there's still an issue.

closed 2020-05-17 by hydra